nodemcu/nodemcu-firmware #1268: DHCP IP address visible to router, Static IP address not  (Esplorer IDE)

Foxdesign2016, 2016-04-27:
Using Esplorer IDE to program the ESP device for DHCP IP assignment :
wifi.setmode(wifi.STATION)`
wifi.sta.config("SSID","PW")
 works fine (ESP device functions as a web server and I can see it on my routers list of devices) but changing it to a static IP address by adding this: 
wifi.sta.setip({ip=192.168.1.xx,netmask="255.255.255.0",gateway="192.168.1.1"})
 The web server is still available on the ESP device, but the router doesn't see it anymore.

 Is this a known firmware issue or possibly an IDE issue? currently, by using a static IP you cannot pipe data to the esp device from another device on the network.

dnc40085, 2016-04-27:
is this a duplicate?

Foxdesign2016, 2016-04-27:
yes sorry- was trying to edit into the correct format

closed 2016-04-28 by marcelstoer