nodemcu/nodemcu-firmware #1269: Support for ESP8285?

marcelstoer, 2016-04-27:
Can we support ESP8285 with little effort?

https://espressif.com/en/media_overview/news/espressif-announces-esp8285-wi-fi-chip-wearable-devices

http://imgur.com/M0uVrqT

vowstar, 2016-04-29:
ESP8285 is same as ESP8266, and only use DOUT mode. So we should change esptool.py to DOUT by default. NodeMCU dev0.9, 1.0, ESP-12, ESP-12-E, ESP-12-F all support DOUT.

marcelstoer, 2016-06-15:
Just found this chip in the wild on a nice dev board at https://www.tindie.com/products/onehorse/esp8285-development-board/.

eku, 2016-06-16:
@marcelstoer nice board, but expensive.

marcelstoer, 2016-06-23:
The board is real nice and NodeMCU just works out-of-the-box: http://frightanic.com/iot/nodemcu-esp8285-perfect-match/ .

closed 2016-06-23 by marcelstoer