nodemcu/nodemcu-firmware #1276: UART bitcount problem using parity and multiple stopbits

TheoHarbers, 2016-05-01:
I have a problem to configure UART to read datastream at 1200 bps, 7, EVEN, 2 stopbits

uart.setup( 0,1200,7,uart.PARITY_EVEN,uart.STOPBITS_2,0) gives a lot of errors in the data

uart.setup(0,1200,7,uart.PARITY_NONE,uart.STOPBITS_2,0) works but of course parity is ignored
uart.setup(0,1200,7,uart.PARITY_EVEN,uart.STOPBITS_1,0) works with one stop bit less

It looks like the total bitcount is wrong. the datastream is 7+parity+2 stop = 10 bits, the uart.setup works  when i setup it up for a total of 9 bits.

Does anyone else have this problem?

FrankX0, 2016-05-22:
Cannot reproduce what you are reporting.
At least with the latest dev build and tested with RealTerm it works fine.
Could the problem be on the transmitter side?

TheoHarbers, 2016-05-22:
I have not tried it with the latest build but for me but the 'problem' is solved just using 1 stop bit. For me it was just a matter of sharing this with others and maybe someone experienced the same thing.

closed 2016-05-22 by TheoHarbers