nodemcu/nodemcu-firmware #1280: Math library

Alex-R-A, 2016-05-02:
So, I am missing some basic features like math.floor(), square root, etc. Can I start on "math" module to be added to master? Or there are other solutions?

pjsg, 2016-05-02:
Note that we only take PRs targeted at 'dev'.

Alex-R-A, 2016-05-02:
My bad, dev, that is.

Alex-R-A, 2016-05-02:
PS: How do you test the code? Build the image then upload it to esp8266? Or is there an easier way?

pjsg, 2016-05-02:
I build locally and then flash the code and then run some tests. For the math functions which don't have any external dependencies, it should be possible to write a bunch of tests in lua and then just execute them. There will be a problem that the platform doesn't handle big lua files (owing to the restricted memory), but it is probably better to have more smaller files.

Adding the tests to the repo would be very nice.....

One thing to note is that there are float and integer builds and many of the math functions do not make sense in the integer build.

vcchm, 2017-02-18:
I found two solutions for this problem
   i) add "ICACHE_FLASH_ATTR" before any trigonometric function in lmathlib.c
   ii) (ugly one - will solve your problem but out of scope of this forum) calculate sin & cos in LUA. I did this when I did not want to rebuild a firmware

function fsin(x)
  local p=-x*x*x
  local f=6
  local r=x+p/f
  for j=1,60 do
    p=-p*x*x
    f=f*2*(j+1)*(j+j+3)
    r=r+p/f
  end
  return(r)
end

function fcos(x)
 return fsin(PID-x)
end

vsky279, 2017-03-12:
I needed trigonometric functions for calculation of sunrise/sunset times. I have implemented it very simply by storing pre-calculated results of sin and asin functions. With double precision, 256 equal intervals and linear interpolation I receive sufficient precision. Sunrise/sunset times differ by max 5 seconds from precise calculations. And it's very quick.

The source is here:
https://github.com/vsky279/nodemcu-firmware/blob/sun/app/modules/sun.c
https://github.com/vsky279/nodemcu-firmware/blob/sun/app/modules/sun.h

It was more a workaround approach as I did not manage to make the trigonometric functions in math lib to work.

A natural logarithm is implemented in the BME280 module (https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/bme280.c#L208).

vcchm, 2017-03-12:
Using precalculation only makes sense if you need  speed (for gaming). If not my routine will do the jobs with more precision...

vsky279, 2017-03-12:
By your routine do you mean the lua function `fsin` quoted above?
The function does not define `xxx`, `pxx` and `f2`?? I don't know this approximation. Is it somewhere documented?

vcchm, 2017-03-12:
const PID = pi / 2

function fsin(x)
  local p=-x * x * x
  local f=6
  local r=x+p/f
  for j=1,60 do
    p=-p * x * x
    f=f * 2 * (j+1)*(j+j+3)
    r=r+p/f
  end
  return(r)
end

function fcos(x)
 return fsin(PID-x)
end

vcchm, 2017-03-12:
This approximation is just taylor series approximation of sin, and cos (x) = sin(PI/2-x) rule

vsky279, 2017-03-12:
Oh, sure.
[image]

If the application was running on batteries than a lookup table might probably be better. Otherwise I agree Taylor is better.

marcelstoer, 2017-06-21:
Related: espressif/ESP8266_NONOS_SDK#16

Kielkopf, 2017-09-26:
Ich hab das ganze noch etwas "optimiert um es bei mir einzusetzen.
Wenn man das berechnete Intervall auf 0-90° begrenzt, kann man schon mit wenigen Iterationen ein relativ genaues Ergebnis erzielen (hier mit dem for bis 10 sind das ca. 5-6 Stellen hinter dem Komma)

wer mehr Genauigkeit braucht, kann einfach die Zahl 10 durch eine höhere ersetzen. Mehr als 20 bringt anscheinend nichts mehr.

```
function fsin(s)
  local x=s  
  if x>180.0 then
    x=(x+180)%360-180
  elseif x<-180.0 then
    x=(x-180)%-360+180
  end
  if x>90.0 then
     x=180.0-x
  elseif x<-90.0 then
     x=-180-x
  end
--  print (x)
  x=math.pi*x/180.0
  local p=x 
  local f=1
  local r=p
-- print (x)
 -- 6 Stellen hinter dem Komma fuer 1/4-Kreis
  for j=2,10,2 do
    p=-p * x * x
    f=f*j*(j+1)
    r=r+p/f
  end
 return(r)
end

function fcos(c)
  return fsin(90-c)
end
-- TEST:
for w=-360,360,1 do
     print (1-fsin(w)*fsin(w)-fcos(w)*fcos(w))
end
```
P.S. Ich hoffe ich hab keinen Bug drin, bin Lua-Neuling

closed 2018-02-24 by marcelstoer