nodemcu/nodemcu-firmware #1287: BME280 module humidity reading

TheoHarbers, 2016-05-10:
BME280 sensor humidity readout gives very different values then on RaspberryPi using the Adafruit_Python_BME280 library using the same BME280 sensor in a closed box protected environment. 

NODEMCU:    {"temperature":23.4,"humidity":57.3,"truepressure":996.5}
RASPBERRY:  {"temperature":23.4,"humidity":45.9,"truepressure":996.3}

humidity 57.3 - 45.9 = 11.4 difference

Also pressure readout is very different compared to BMP085 (in the same box)
BMP085 on NodeMCU: {"temperature":23.4,"truepressure":993.4}

pressure difference between BME280 and BMP085 = 996.5 - 993.4 = 3.1 hPa

Could there be any mistake in the many calculations for the BME280?

The BMP085 reads out the best pressure compared to my analog pressure meter and other weather station.

dvv, 2016-05-11:
Can you please try https://github.com/dvv/nodemcu-thingies/blob/master/bmp085.lua on the same setup?
TIA

TheoHarbers, 2016-05-11:
I have the BMP085 working just fine for a long time it is just the new BME280 that show strange readings. But i can test the lua BMP085 if you think it helps to understand this strange behaviour. I also ordered a couple more BME280 to see if they show the same values.

vsky279, 2016-05-19:
Are you using the BME280 module which is in the dev version?

If you are using only some Lua version of compensation formula calculation then there is very likely some imprecision in compensation calculation. It is quite difficult to implement 32bit floating point calculation in Lua.

I did it (before realizing it is much, much easier to calculate it in c) and you can test the precise compensation calculation using these Lua modules: https://github.com/vsky279/BME280.

vsky279, 2016-07-24:
As there's no discussion I would suggest to close this issue.

devsaurus, 2016-07-24:
I agree, thanks @vsky279.

closed 2016-07-24 by devsaurus