nodemcu/nodemcu-firmware #1288: Some WiFi functions may contribute to premature flash failure

dnc40085, 2016-05-11:
The problem here is that the WiFi functions `wifi.setmode()`, `wifi.sta.config()` and `wifi.ap.config()` all save their configuration to the flash by default. If an application developer writes a program that configures the WiFi at startup, that means that on **every single startup** the flash written to at least once.

Luckily, there are some functions that were recently added to the SDK that allow for this functionality to be optional (`wifi_set_opmode_current`,  `wifi_station_set_config_current`, `wifi_softap_set_config_current`).

I'm also planning on exposing the following related functions:
* `wifi_get_opmode_default` (get wifi mode stored in flash)
* `wifi_station_get_config_default` (get station configuration stored in flash)
* `wifi_station_ap_number_set` (set # of APs that will be stored in flash)
* `wifi_station_get_ap_info`(get list of APs saved in flash)
* `wifi_station_get_current_ap_id`(returns index of current AP from list saved in flash)
* `wifi_station_ap_change`(connect to alternate AP saved in flash)
* `wifi_softap_get_config_default` (get softap config stored in flash)

The purpose for this issue is to get an opinion on what the default behavior and what the names should be for these functions, before I write a PR to add this functionality.

jmattsson, 2016-05-11:
I'm hesitant to change the default behaviour, but I could totally get behind an implementation change which only calls the SDK functions if the new config differs from the current or saved config.

dnc40085, 2016-05-11:
I like your idea, it seems more noob proof than just changing the default behavior.

marcelstoer, 2016-05-12:
> I think the new default behavior should be **not** to store the configuration in flash.

I disagree and would much rather see Johny's idea implemented.

jmattsson, 2016-05-14:
> means that on every single startup the flash written to at least once.

Btw, has this actually been verified? I thought I'd seen those functions not touch the flash if the new data was the same as the old. Would save us some trouble if the SDK already does the smart thing here.

dnc40085, 2016-05-14:
>Btw, has this actually been verified?

 I'm not sure about when you run `wifi.sta.config` with the same credentials,
 but after executing `wifi_station_ap_number_set(5)` to increase # of saved APs from 1(default) to 5 then running `wifi.sta.config()` with alternating credentials, then checking stored APs with `wifi_station_get_ap_info`, I get the following:
```
index: ssid:                  password:                      bssid:            
0        test1                  password                      ff:ff:ff:ff:ff:ff 
1        test2                  password                      ff:ff:ff:ff:ff:ff 
2        test1                  password                      ff:ff:ff:ff:ff:ff 
3        test2                  password                      ff:ff:ff:ff:ff:ff 
4        test1                  password                      ff:ff:ff:ff:ff:ff 
```

Aside from hooking up a logic analyzer or trying decipher the asm source, is there anyway to confirm whether or not the credentials get overwritten when trying to set them to the same values?

pjsg, 2016-05-14:
How about doing some accurate timing -- if there is a timing difference between 'same creds' and 'different creds' then that suggests that the optimization is already present////

dnc40085, 2016-05-14:
Good idea, I'll give it a try!

closed 2016-11-07 by dnc40085

dnc40085, 2016-11-07:
the fix(PR #1497) for this issue has been merged

PR #1497 (2016-11-06): Update to the wifi module
Fixes #1288 and #1480 and #1237 
- [x] This PR is compliant with the [contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

This PR adds additional functionality to the WiFi module, the changes made are described below.

Functions Added:
- [wifi.getdefaultmode()](https://github.com/dnc40085/nodemcu-firmware/blob/dev_wifi_update_2/docs/en/modules/wifi.md#wifigetdefaultmode): re [... cut]