nodemcu/nodemcu-firmware #1289: Idea to increase available RAM -- may be crazy

pjsg, 2016-05-11:
One of the significant consumers of RAM on the nodemcu is the compiled code and associated constants etc. This can be loaded from disk (if the .lc file is stored there).

Once loaded this code (and associated data) is (I think) constant and never modified. This means that it could be stuck into flash, thereby saving a bunch of RAM (and also enabling larger applications).

There seems to be a number of possible approaches: 

- When luaU_undump loads the .lc file, after loading, copy all the code and constant data into a special flash area. In order not to wear out the flash, first check to see if the data block is already stored there.

- Have a way to load a .lc file into the flash and then use the standard load technology to make it into a runnable object. 

It turns out that there is already some support for mapping readonly objects into memory, though I don't know what it was/is used for. You would need to fix the memory free function to ignore freeing pieces of the flash.

In the first case, you would need to have a way of clearing out the flash when it becomes full of junk. The easiest way might be to have a function that sets a flag that causes it to be reset on the next boot (before anything gets loaded into it). It might also be possible to be smarter about setting the flag (e.g. set the flag automatically if there was insufficient space, and that the flash area contained data at platform boot)

In the second case, you would also need a reset method on the area, and then you would need to reload all the files that you wanted. 

I don't think that you can point into the spiffs filesystem as it does wear levelling and might move blocks around.....

rm3nchaca, 2016-05-11:
It sounds really interesting!

jmattsson, 2016-05-11:
Terry and I bandied that very idea around quite some time ago, but didn't come up with any workable solutions. If the .lc files were actually stored with sequential data it would be "easy" to add support for running the byte code straight out of flash, but that's not the case. Maybe `node.compile()` could be changed to save into a dedicated code area, but then we get all sorts of interesting questions on how we make that area visible (so users can erase unwanted things, at least), how to allocate the area in the first place (especially on small flash chips), and of course all the usual pitfalls of fragmentation you get with any memory allocator. It's an area riddled with pits with very sharp things in them...

dnc40085, 2016-05-27:
This idea sounds similar to [flashmod](http://www.esp8266.com/viewtopic.php?f=19&t=1940), at one point I attempted to port it over to C but couldn't figure out how to pull it off.

TerryE, 2016-05-27:
PHP uses a similar strategy in Opcache, etc. But as Johny says if you can't guaranty that the storage is contiguous then it gets v messy. The other thing is that the true RO bits of the code are quite small say 50%, and you really need to do something similar to the rotab patch to increase this. Not a trivial thing to contemplate. I rather spend my time porting net to LwIP.

pjsg, 2016-05-27:
It nearly works... but there are some strangeness which happen in complex cases. I think that there is an interaction with some interrupt handlers. 
---
Philip Gladstone. On the road, please excuse typos.

pjsg, 2016-05-28:
I suspect that interrupt handlers are not allowed to access flash memory for data (without possibly causing problems). Does anybody know if this is really true?

Are there any interrupt handlers left in the nodemcu code that might access lua objects? The only thing that slightly worries me is that the prompt for the interactive lua -- the "> " is readonly. It is printed with a c_puts() and I don't know whether this is buffered before being sent to the UART. I expect that there is an interrupt handler somewhere that outputs the next character -- is this in the nodemcu code or somewhere else?

Thanks

jmattsson, 2016-05-29:
An ISR cannot know whether the flash memory is mapped or not when it gets executed, which is why all the ISR code needs to sit in IRAM (because executing the ISR from unmapped flash gives predictably crashy results!). Oddly enough the SDK notes for the RTOS SDK states that ISRs may now reside in flash; I have no idea why they would claim that and whether it holds true for all the things we do - I haven't gotten that far with my RTOS work yet. Still, keeping anything an ISR needs in \[I\]RAM is a good idea if you ask me.

As for the UART, yes, that is all buffered.

TerryE, 2016-05-29:
Johny beat me to this.  Don't know about RTOS yet, but +1 on other ISR comments.

TerryE, 2016-05-29:
Philip, if you are serious at exploring this then you should look at my LCD patch whitepaper.  Doing full path testing of this sort of patch on eLua without a full test suite just doesn't give you the sort of coverage that you need.  I developed the compact debug patch under bog standard Lua on a PC dev environment first.  You could simulate flash by using a shared memory region and mapping it in R/W and R/O as required.  The Lua test suite through up all sorts of issues that I would never have discovered if developing on the ESP platform only.  Just a thought.

Have you done any rough-cut writeup on what elements you are trying to access from ROM and what you are deep copying into RAM?  The 32 bit code **Instruction** vector is PIC and R/O, ditto the **packedLineInfo** vector, and these could be accessed in Flash.  But pretty much everything else is is non-PIC or has variable elements in it IIRC and would need to be in RAM.

pjsg, 2016-10-03:
My code now works quite well. The basic thing is to copy strings into flash and the program code and sometimes constants. A little bit of change was needed to work around having two different strings with the same value now be equal (one in flash and one in ram). 

There is still a fair amount of ram being used in the application that I testing this on (a humidity & light controller for a carnivorous plant tank), and I'm not sure what it is. 

I'm also a bit concerned that building with the node DBG enabled now causes problems. I'm wondering if the stack consumed by c_printf has grown in the newer SDKs. It turns out that my concerns about ISR accessing flash were all bogus, and what was actually happening was that the NODE_DBG calls were pushing the stack sufficiently deep that it runs into the UartDev block and corrupts the buffer pointers.... Bad things then happen. Adding more debugging printfs makes things worse!

TerryE, 2016-10-03:
Phil, strings are idempotent in Lua, so there is no point in having two copies: one in Flash and one in RAM.  Stings are accessed through a simple hash table `lstate.h:stringtable` with one entry per live string.  The hash is computed from the string value, but the comparison itself is based on the gc.value address. If you compare the **nodeMCU** and standard **Lua-5.1.5** versions of `lstring.h` then you will see that **eLua** has already added some support for readonly strings stored in ROM (and which **nodeMCU** doesn't use, BTW).  

The VM assumes that strings are carbage-collectable objects, so you need to be careful about moving them into Flash, especially as some Lua code can almost pathologically creates temporary strings which need to be GCed to avoid memory exhaustion.  Again the eLua variant has some limited support for ROM-based readonly strings which prevents their GC (compare the standard and eLua/nodemu versions). 

If we wanted to do anything, then I think that the best way to save RAM here would be to allow the RO elements of the Lua code to be Flash-based, and more specifically in the memory-cached flash window.  But this would be quite a lot of extra work, IMO which will take time and effort.  Wouldn't it just be simpler to wait for ESP32 availability for this class of application? 

However, IMO the main issue with string handling in Lua is that the VM and RTS is optimised for time and simplicity at the cost of space resulting in characteristics that are undesirable on embedded systems.   The two key expects that I would focus on here are:

-  _The **luaL_Buffer** implementation_. Many `lstrlib.c` use this to maintain working string copies, as well as some I/O, table and library routines.  It is internally complicated and a bit bizarre.  The basis is that a working fixed-length buffer (of size `LUAL_BUFFERSIZE`) is allocated on the stack and this can only be filled by appending bytes.  Once full, the contents are flushed to a new Lua string that is `LUAL_BUFFERSIZE` long and appended to the Lua stack, and the buffer is emptied. Once the processing is complete, any buffer residue is pushed to a new Lua string and these string chunks are concatenated.  Because the concatenate operation short circuits the case of one string:
  -  This all works out very efficient so long as the string is less than  `LUAL_BUFFERSIZE` bytes long
  -  It works fine for longer strings albeit at the cost of creating temporary  `LUAL_BUFFERSIZE` sized chunks which are no longer referenced and therefore can be garbage collected.

On virtually all Lua targets, the memory overhead of maybe a dozen functions on the C stack using this **luaL_Buffer** implementation is 12Kb and this is a trivially acceptable overhead for a simple, fast scalable strategy that is fast for the vast majority of lua strings.  However unfortunately this isn't the case on the ESP8266. 

- _Some string routines could trivially determine a known size for the target string_.   For these, it makes sense to have an alternative `alloca()`, based strategy.  We could either make this an alternative strategy for routines such as  `str_reverse`, `str_lower`, `str_upper`, `str_rep`, `str_char` or simple extend the  `luaL_Buffer` always to use an alloca-based buffer on a specified length and base all string chunks on this initial size.  Even if we can't easily compute the known size, in many cases we could still estimate a good estimate that will be significantly less than the default (`BUFSIZ` =1024), and still avoid 90%+ of the tempoary chunk string creation.  And I would also suggest that 128 is a more appropriate default where we don't estimate the size for most ESP applications.

Sorry to be so verbose, but I think that this is worth an explanation :smile:

pjsg, 2016-10-03:
I realize that strings are immutable in Lua. The existing code doesn't always handle the READONLY flag on strings correctly (in particular when printing error messages).

In any case, lua code contains strings which are only found in that source -- many print calls have unique strings. It was easier to save them all into flash rather than try and work out which ones can safely be left in RAM. 

I also didn't want to leave the readonly TString in RAM either as that consumes 20 bytes plus the memory allocation overhead. 

The goal of this work is to make it possible to have larger lua applications running. The httpserver implementation that I use goes to great length (in the original code) to dynamically load modules to handle each request (in order to keep its memory footprint small). This has a severe performance impact. By keeping those modules effectively in flash, and loading them once, it now uses less RAM and is much faster.

pjsg, 2016-10-03:
It looks to me as though nearly all of the luaL_buffer stuff handles small buffers correctly. One of the uses in liolib.c does not and hence would need some work. I will probably create another ticket to do the buffer work..... 

There is definitely a tradeoff between memory usage and performance -- but it is probably worth changing the buffer size from 1024 to (say) 128.....

TerryE, 2016-10-03:
Philip, re string handling, and minimal code sizing, I feel that what you are really looking for is something more akin to the PHP APC or OPcache architecture; in this the RO elements of the compiled code are copied into an SMA and any R/W elements are deep copied into RAM.  For Flash-based compiled code to be effectively usable, the code has to be within the Flash memory region that can cached into RAM.  I think it would be dangerous to assume that the entire filesystem (and now VFS) can be memory mapped.  Having to statically bind code into your build would be a real pain.  I feal that if we are going down this route we should have an architecture which reserves a region in directly addressable Flash for Lua code, and a new `string.dumpToFlash()` method which can be used to compile file-based `.lua` into this region.  This approach would envisage doing a special initialisation pass, to reinitialise the Lua flash, but this has three upsides:

1.  This greatly simplifies compile and test cycles, since the Lua flash can be reinitialised with having to re-flash the entire firmware.  The `dumpToFlash` method would handle any address resolution so that pointers in the compiled data structures have the correct addresses.

2.  Using a `string.dump()` variant means that the flash initialisation pass code also compile generated code, so that configuration-specific data structures, etc. could be compiled to Flash.

3.  You would still need to "load" flashed code into scope, but this would be using a flash-based direct addressing, so deep copies of the R/W elements if any (need to thing about these) would be _fast_.

Anyway,  that's my 2 cents worth :smile:

TerryE, 2016-12-01:
@pjsg Philip, if you want we can reissue the following as a separate issue.  I've been thinking more about this proper support of Lua ROM objects.  Forgive me as you probably know the background, but I've added this for the wider readership.

## Background

-  In Lua all strings are [Interned](https://en.wikipedia.org/wiki/String_interning), that is they are immutable and unique.  So there is no method of modifying a string; you can only create new strings.
-  The standard Lua type TString is used to contain a string.  This contains a standard GC header, the strings hash, value and the string value.
-  Equality and assignment of strings is cheap.  Since strings are unique, two string values are the same if and only if they point to the same TString object, so a string compare becomes an (aligned) address comparison.
-  Creating strings is expensive, in that its hash has to be computed and used to lookup up the string in `GL->strt`. If it already exists in the table then the address of the existing TString is returned, otherwise a new object is created and entered into the hash.
-  Accessing string content is comparable to standard C string manipulation, though some short circuits are possible as the length of the string is already known. 
-  Lua string are just standard Lua objects that can be GCed just like any other proper Lua object, and in the case of intermediate string temporaries resulting from concatenation, then this will be accomplished in the next two GC cycles.  (I often see patterns `resp=a..b..c..d..e` and this will generate a _lot_ of temporary strings to be freed during subsequent GC).
-  The NodeMCU/eLua patch has a special additional case for ROM strings, in that the TString contains a pointer to the ROM object instead of a copy in RAM, but the TString itself is still in RAM and referenced in the `GL->strt` hash table.  These RAM objects can still be subsequently GCed.

## Observations

-  Even though the library code uses ROM strings, using these creates a RAM footprint for the Tstring objects, the GL->strt entries, and (since the cost of GC is proportional to the number of objects) a GC performance cost.
-  Most ROM library code converts Lua-supplied TString variables and compares them against C string constants using the string library primitive like `strcmp`.  This is expensive, and doubly so since ROM strings accessed by byte instructions require handling by a software exception handler.
-  Wouldn't make a lot of sense to offer the C-based library developer a set of macros/ primitive to declare ROM-based TStrings and use TString based comparison / assignment / manipulation where practical? 

## Outline Proposal

-  We add the ability to include source declaration of ROM TStrings, for example: `LUA_TSTRING( net_pOpen, "open" )` which will result in a `extern Tstring *net_pOpen` being resolved at link time to the ROM TString for "open".
-  ROM Lua Objects are not linked into the GC chain and are excluded from GC
-  A process pass is part of the make which collects LUA_TSTRING declarations and build a module statically declaring the objects and a new hash table which will be bound during initialisation to `GL->ROMstrt`.
-  The function `lstring.c:luaS_newlstr_helper()` which include a second for loop to check the ROMstrt before calling `newlstr()`
-  Note that multiple declarations of the same constant will be folded into the same ROMTString object but with multiple externs.  
-  Also note that any Lua code using these ROM strings will inherit the ROM copy.

Lastly note that in principle this approach could be extended to include other Lua ROMable types such as prototypes.

jmattsson, 2016-12-02:
Here's a little something that might be useful as part of the toolkit to achieve this. The other night when I was chatting with @TerryE I discussed the possibility of using a wee bit of C++ to do *compile-time* calculation of the Lua hash values to avoid having to do a relink.

```c++
template<const char *str, size_t l>
class lua_str_hash
{
  static const uint32_t seed = l;
  static const size_t step = (l >> 5) + 1;

  template<bool more, size_t l1, uint32_t h>
  struct compute
  {
    static const uint32_t value =
      compute<(l1 < step), l1 - step, h^((h<<5)+(h>>2)+uint8_t(str[l1-1]))>::value;
  };
  template<size_t l1, uint32_t h>
  struct compute<false, l1, h>
  {
    static const uint32_t value = h;
  };

public:
  static const uint32_t value = compute<true, l, seed>::value;
};

// Usage: note that the strings have to be "constexpr" with external linkage
constexpr const char hello[] = "Hello";

// Not a TString, but should get the idea across I hope
struct my_str_t
{
  const char *str;
  uint32_t hashval;
};

const my_str_t my_str = { hello, lua_str_hash2<hello, sizeof(hello) -1>::value };

printf("%s => %u\n", my_str.str, my_str.hashval);

```

Add a bit of C pre-processor macros to wrap it up nicely, and perhaps look at whether the global strings could be named deterministically and marked weak, to hopefully get the linker to fold any duplicates together. Worst case we could have a single file which declares ROM strings, I guess.

TerryE, 2016-12-02:
@jmattsson, yup I StackOverflowed this one as well after we talked. We could use this technique to generate TStrings in CC+. However I couldn't think of a way to get the linker to fold duplicates to make them unique.  We can avoid the two pass link by preprocessing the source or compiler outputs. Need to think about elegant implementations.

One advantage of this TString approach is that we don't to do a major rewrite of rotable as the list of string pointers becomes a list of unique TString \* so no need to do strcmp calls, just the pointer compare for equality.

pjsg, 2016-12-02:
In my current implementation, I took the opposite approach (as it was simpler). Strings are no longer uniquely defined by their pointers -- if the pointer is into RAM, then it is still true, but if into flash, then it might be duplicated in the RAM. There were only a couple of places where string comparison for equality was performed, and I just changed those. This approach definitely is not as clean as the methods above.

One of the problems that I faced was how to maintain any sort of lookup table into the flash area. I didn't want to build any large RAM data structures to index into it. This results in having the writing process be fairly inefficient (it does a linear search over all the objects), but it is light on the RAM usage.

TerryE, 2016-12-02:
@pjsg, thanks I understand your approach.  What I am suggesting is that we use a Lua string preprocessing step that essentially generate an assembler file containing all of the ROMTString static declarations including initialisation and the corresponding external definitions so that the external references in the library code can be correctly resolved at link time.   Ditto ROMstrt hash table.  There's about a 5 line change to `lstring.c:luaS_newlstr_helper()` and striping out some of `strcmp()` calls and replacing them with pointer comparisons.  I think that this discussion is worth bouncing around a bit more.

pjsg, 2016-12-02:
I did something like this for the improved ROTable lookup patch. This postprocesses the built image, extracts strings, computes hashes and then relinks with a dynamically generated C file. Fancy use of weak symbols makes it all work.

TerryE, 2017-06-26:
Going back to Phil's original post:

> There seems to be a number of possible approaches:
> -  When luaU_undump loads the .lc file, after loading, copy all the code and constant data into a special flash area. In order not to wear out the [...]

and a point that Johny made on our side emails on the Lua 5.3 port:

>  I haven't heard many (if any) complaints about running out of flash storage, it's always RAM, RAM and RAM.

I've been brooding some more about this.  My first assertion is that given the Lua 5.3 port already supports ROMable tables and strings efficiently and is progressing steadily, we should close this issue as far as Lua 5.1 is concerned as there's just not point in trying to take this further for 5.1.  However in the context of the future 5.3, I think that we could make major improvements here, and these would be along the line of Phil's original suggestion.  What I suggest is that 

-  The build allocates a configurable "partition" (which would probably be a real partition in the case of ESP32 builds) that is flash sector aligned and an integral number of flash sectors (e.g. 64Kb) and within the program flash addressable window.

-  We have (i) a host-based (a luac option) for building a "Lua Readonly partition" (***LROP**) image which can them be overlaid onto the partition using **esptool.py**, and (ii) given that many Lua developers don't have a firmware build environment, a ESP-base build process for building and storing the a partition image.

-  This partition wouldn't be a filesystem as such, but just a block of directly addressable RO memory that is writeable on a one-pass update process.

-  As an aside and for my sins prior to my involvement in Lua, my previous OS project was support the PHP internals and mainly the Opcache system.  This allowed PHP compiled code to be stored in shared (and effectively RO memory.  The main trick was that requiring a module would reference RO data structures and the RW data-structures would be deep copied into process RAM.  This whole approach has a direct equivalence in the case of LROP.

-  Because of interoperation with SPIFFS and other Lua library functions, I see the ESP-based process is a special boot mode which uses the ESP in a dedicated manner to process a list of supplied modules and generates a new LROP and then restarts the chip.

-  The RO elements of LROP based modules would be addressed directly in the partition address space, and specifically each functions `Proto` structure and all its linked data hierarchy: the code, the constants including striings, any debug meta-data such as line info.  (Note thaet Lua  5.3 introduces one R/W field in the `Proto` structure, `cache` but dealing with this is straightforward.)

-  The `LClosure` structures would still need to be created in RAM dynamically as these and their linked dependent data hierarchy are RW.  

-  So the broad consequence is that the code and constant elements of LROP would be eexecuted directly out of RO flash and only the RW data elements out of RAM.
    -  **This would effectively double the usable RAM** if the bulk of code was LROP-based modules
    -  Loading of LROP modules would be ***fast*** as the only load code would be to set up the outer closures.

-  The only slightly constraint is a result of the RO string table implementation, which mirrors the RW one.  Here strings are hash and linked chains based on its modulus on the hash table size.  The RW implementation uses a size of the smallest  2^m which is larger than the number of strings, which give an average chain length of <1.  So if the application has 600 active strings between GCs then this table will be 4K long.  There is nothing magical about this.  For example it could be half the size with  the cost of lookup being say 2 compares on average instead of one, but the size of the RAM hash vector is adaptive.  In the case of ROStrings the current 5.3 implementation needs to address them directly, and so the hash table size has to be fixed because the link chains are calculated using this.  However, if we move the into the start of the LROP and threefore fix the address at link time, but not the content, we can make the generation and  sizing of the string table part of the LROP build process.    
 
Whether the LROP was generated on host or on ESP, the modules in the LROP would be in the `package.searchers` path and so would be loaded automatically by a `require`.   I already use this technique for my own Lua searcher to dynamically load modules over the network if they are newer than the SPIFFS version.  This makes developing so much easier when you just edit the Lua file on the host and restart the ESP to refresh the SPIFS copy automatically on the fly.  There nothing to use this approach in the LROP rebuild process.

So in summary, everyone complains about the lack of RAM when they use Lua.  Yet we load the Lua code from RAM even when the rest of the RTS runs from Flash.  This is crazy.  Why do this?  Being able to run Lua modules direct from Flash would reduce load times and pretty much double available RAM.   We can failry easily add this to the Lua 5.3 implementation.

@pjsg, @jmattsson, @devsaurus, @djphoenix, @marcelstoer comments?

pjsg, 2017-06-26:
I have a version of the code for 5.1 that appears to work in my test cases (though it is now based on a fairly old dev branch). https://github.com/nodemcu/nodemcu-firmware/compare/master...pjsg:readonly-code There are some unrelated changes there, but the code really boils down to one file: freezer.c and some minor changes to the core lua code to make it work.

Because my code did not support ROStrings in a native way, I had to change the way that string equality checks worked.

The other serious nasty issue was the read-only flash caching. If the processor reads some data from the flash, then you write to the flash, and then you read again, you will get the old value. I found that I needed to read other areas of the flash to trigger the cache flush. See lines 196 & below from freezer.c

TerryE, 2017-06-26:
> I have a version of the code for 5.1 that appears to work in my test cases (though it is now based on a fairly old dev branch).

Phil, if you want to put this through rigorous testing and release it to dev then that one option for you.  However,  I am doing mine as part of my Lua **5.3.4** port, which has really proved to be a major task, because I am trying to set out project on a firm and stable Lua base for the next few years, and I don't want to get distracted at all by Lua 5.1 branches.

However our thinking has often come to the same conclusions and observations.  I did have a a quick lok at oyur stuff, but without any form of documentation or write-up I found it difficult to be my head around it.  I do need to go through this because some of the low level issues that you have solved and some of your implement might be reusable.

We need to think of the use cases.  Marcel has had over 50,000 different users download flash images and this is probably at least two order of magnitude larger than the number of users who are capable or willing of doing their own local builds.  Hence I see the ability to build what I call LROP images with just a standard firmware build and possibly a (Windows runnable) host-based python or Lua script in support as an essential option.

> The other serious nasty issue was the read-only flash caching. If the processor reads some data from the flash, then you write to the flash, and then you read again, you will get the old value. I found that I needed to read other areas of the flash to trigger the cache flush.

Thanks for the reminder as it is an important point, but I had already though about this one and propose a kiss approach.  My suggested LROP rebuild uses RAM-based lua code loaded from SPIFFS or over the networks and some C based helper routines, and reboots the ESP on completion.  The area where I need to be carefully is the overwrite of the ROstrt and ROString resources used by the RTL so this will be an epilogue bootstrap: the code that rebuild the string table will not depend on these RO resources and will issue a `system_restart()` before longjumping control back to the SDK.

pjsg, 2017-06-27:
I've just added a bit of documentation in my branch (and backmerged dev) so that the diff is now sensible: https://github.com/nodemcu/nodemcu-firmware/compare/dev...pjsg:readonly-code

I'd rather not merge it into the 5.1 codebase because of the difference between that model and the model that makes sense in 5.3. (i.e. I don't want to have to change the API).

My current approach is pretty transparent to the developer. Just by adding a definition like:

```
dofile = function(f)
    return freezer.freeze(loadfile(f))()
end
```

they can get all their content copied into flash. The bad news is that the management of this is hidden so that you can't guarantee (depending on the state of the flash) how much RAM will be saved.

TerryE, 2017-06-27:
Lua goes to great length to abstract `require` pathing and sources from the developer.  The loader maintains a list of `searchers` (in Lua 5.3 speak, though the equivalent functionality is in 5.1  Basically the developer (or the RTS startup) can add or modify the searchers.  I use this with my own searcher so that doing a `require "modX" will load a new version over the network (and update the SPIFFS copy) if I have updated the master on my dev PC.  I don't have to think about or use the tedious ESPlorer downloader.  So if we go this way then `require "modX"` would use the LROP version if it existed.

As I said, I would prefer to go down the rebuild LROP route because this enables all of the code, metadata and strings to be moved into flash so that only the true RW stuff ends up in RAM.   It also discourages dumb flask churn which will limit the life on a module.

pjsg, 2017-06-27:
I think that we are in basic agreement. The only issue is how to load the partition with code. Mine is an incremental approach, and yours is a batch approach. Figuring out what the exact test/debug/reload cycle consists of is important for developers. I like your approach of being able to load source files over the network.

TerryE, 2017-06-27:
Philip,  one of the main advantages with the batch method is my approach to strings: as well as Lua standard RAM sting table, I have another in RO Flash and any strings are first resolved against the RO string table, and so any use of these strings in Lua applications doesn't take up RAM.  Because Lua functions typically use a lot of strings (particularly if you wan to retain the local and upvalue debug metadata), storing a function in LROP means that you have to rebuild the ROstrt, and I don't want to rebuild this every incremental store. 

As to the increment load stuff, it isn't really a separate module as such, more a snippet that we should make available somehow.

TerryE, 2019-04-10:
Implemented in LFS patch

closed 2019-04-10 by TerryE