nodemcu/nodemcu-firmware #1292: CoAP smoke test

dvv, 2016-05-13:
Hello!

I compiled the following from http://nodemcu.readthedocs.io/en/dev/en/modules/coap/ and tried to get a variable and invoke a function:

```Lua
cc = coap.Client()
--
cs = coap.Server()
cs:listen(5683)
--
state = 0xba55
cs:var("state")
--
function handler(e)
  print(node.heap(), e)
  return "456"
end
cs:func("handler")
--
print(cc:get(coap.CON, "coap://192.168.1.220:5683/v1/v/state"))
print(cc:post(coap.CON, "coap://192.168.1.220:5683/v1/f/handler", "123"))
```

No output to console which means at least that `handler` is not even called.
What am I doing wrong? NB: I do believe it is a _firmware_ question.

Build:
```
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 583ea8514fd454f623a898764b36ca43387d0bcd
	SSL: false
	modules: adc,apa102,bit,bmp085,cjson,coap,crypto,dht,encoder,enduser_setup,file,gpio,i2c,mdns,mqtt,net,node,ow,pwm,rotary,sigma_delta,struct,tmr,u8g,uart,wifi,ws2801,ws2812
 build 	built on: 2016-04-29 08:56
 powered by Lua 5.1.4 on SDK 1.5.1(e67da894)
OK
```

marcelstoer, 2016-05-13:
>  I do believe it is a _firmware_ question.

Made me smile 😄 Do you also believe it's a firmware _bug_?

dvv, 2016-05-14:
Sure. A bug in either _firmware_ or _documentation_. Or both )

closed 2019-06-21 by stale[bot]