nodemcu/nodemcu-firmware #1293: Multiple http.get with individual URLs and callbacks broken or not supported?

marcelstoer, 2016-05-13:
http://stackoverflow.com/q/37218528/131929 confused me and I ran a small test myself.

It appears to me as we don't support calling multiple `http.get` (or other HTTP functions) with individual URLs and callbacks. The below code makes two consecutive `http.get` calls. Each uses a separate URL and a specific callback function.

```lua
function wp(code, data)
    if (code < 0) then
      print("HTTP request to wp.me failed")
    else
      print("wp.me responded with HTTP status " .. code)
    end    
end

function nodemcu(code, data)
    if (code < 0) then
      print("HTTP request to nodemcu.com failed")
    else
      print("nodemcu.com responded with HTTP status " .. code)
    end
end

function run()
    http.get("http://wp.me/pzoQb-sk", nil, wp) -- returns HTTP 301
    http.get("http://nodemcu.com/favicon.png", nil, nodemcu) -- returns HTTP 200
end
```

What I would expect if `run` is invoked manually is "wp.me responded with HTTP status 301" and "nodemcu.com responded with HTTP status 200" in arbitrary order. What I get instead is a total mix-up:

> =run()
> nodemcu.com responded with HTTP status 301
=run()
> nodemcu.com responded with HTTP status 301
=run()
> nodemcu.com responded with HTTP status 200
=run()
> nodemcu.com responded with HTTP status 200
=run()
> nodemcu.com responded with HTTP status 301

This seems related to #1258.

If `run` is invoked automatically in a loop all requests fail:

```lua
for i=1,10 do run() end
```
> HTTP request to wp.me failed
HTTP request to nodemcu.com failed
HTTP request to wp.me failed
HTTP request to nodemcu.com failed
HTTP request to wp.me failed
...

I hope this isn't any embarrassing case where I make a fool of myself because I miss something really obvious.

devsaurus, 2016-05-14:
Quoting the docs:

> Note that it is not possible to execute concurrent HTTP requests using this module. Starting a new request before the previous has completed will result in undefined behavior.

That's in line with the code, since the module can cope with only one single callback at a time.

marcelstoer, 2016-05-14:
> Quoting the docs

Oh boy...I had missed that. I'll update the docs to make that a bit more explicit.

closed 2016-05-14 by marcelstoer