nodemcu/nodemcu-firmware #1298: Passing table to function

Jonathan411, 2016-05-16:
The lua script below generates this LUA Interpretor error:

[image]

`local function tblInsert(tbl, pos, val)`
`    tbl[pos]=val`
`end`

`local function tblGet(tbl,pos)`
`    print("in tblGet: tbl[pos]= ", tbl[pos])`
`    local rtnVal = tbl[pos]`
`    print("in tblGet: rtnVal= ", rtnVal)`
`    return rtnVal`
`end`

`list = {}`
`tblInsert(list,1,"a")`
`tblInsert(list,2,"b")`
`tblInsert(list,3,"c")`

`for i,v in ipairs(list) do`
`    print(i,v)`
`end`

`print("list[1]= ", tblGet(list,1))`

Even though there was a Lua error message the script runs and produces the following output:

`1   a`
`2   b`
`3   c`
`in tblGet: tbl[pos]=    a`
`in tblGet: rtnVal=  nil`
`list[1]=    nil`

I am doing something completely wrong? I have tried 3 different firmware builds from frightanic.com (dev branch) which use Lua 5.1.4 on SK 1.5.1. and always get the same Lua error and script run results. I am running on an ESP8266 ESP-12.

pjsg, 2016-05-16:
This is an issue with esplorer. When you upload a script with esplorer, you must not end a line with a ']' character. The trick is just to parenthesize the expression and then the problem goes away.

Please close the issue if you are using esplorer. If not, then provide the name of the upload tool so that others can learn from your troubles!

Jonathan411, 2016-05-16:
Thanks for the quick response!  I am using esplorer and will close the issue. To solve the problem, i switched to "turbo" mode in esplorer and now it works as expected.

closed 2016-05-16 by Jonathan411

marcelstoer, 2016-05-17:
For reference here's where we discussed this first: http://stackoverflow.com/q/37245889/131929 (and where it should have stay as per http://nodemcu.readthedocs.io/en/dev/en/support/)

Jonathan411, 2016-05-17:
Sorry about that.

 

Jonathan