nodemcu/nodemcu-firmware #1303: Memory leak in socket:on("sent")?

marcelstoer, 2016-05-19:
Do we have a memory leak in `socket:on("sent")` or do I, again, miss something obvious?

Here's my simple test bed of a server that just returns `text/plain`:

```lua
srv=net.createServer(net.TCP)
print("Server created...")
print("Available heap: "..node.heap())
srv:listen(80,function(conn)
    conn:on("receive", function(c,request)
        c:on("sent", function() 
            print("Available heap: "..node.heap())   
            c:close()
        end)
        c:send("HTTP/1.1 200 OK\r\nContent-Type: text/plain\r\n\r\nLED STATE=1\r\n")
    end)
end)
```

Much to my surprise available heap keeps decreasing and never recovers:
> Server created...
Available heap: 42352
Available heap: 43200
Available heap: 43392
Available heap: 42904
Available heap: 42616
Available heap: 42136
Available heap: 41776
Available heap: 41752
Available heap: 41464
Available heap: 41600
Available heap: 41224
Available heap: 41240
Available heap: 40872
Available heap: 40888
Available heap: 40520
Available heap: 40504
Available heap: 40160
Available heap: 39936
Available heap: 39536

To test what impact `socket:on("sent")` has I replaced it with an ugly `tmr.delay()`:
```lua
srv=net.createServer(net.TCP)
print("Server created...")
print("Available heap: "..node.heap())
srv:listen(80,function(conn)
    conn:on("receive", function(c,request)
        c:send("HTTP/1.1 200 OK\r\nContent-Type: text/plain\r\n\r\nLED STATE=1\r\n")
        tmr.delay(50)
        print("Available heap: "..node.heap())
        c:close()
    end)
end)
```
Memory consumption seems to be a lot more stable this way:
> Server created...
Available heap: 42552
Available heap: 42376
Available heap: 42168
Available heap: 41984
Available heap: 41760
Available heap: 41592
Available heap: 41400
Available heap: 41600
Available heap: 41400
Available heap: 41600
Available heap: 40816
Available heap: 41128
Available heap: 40816
Available heap: 41136
Available heap: 40816
Available heap: 41128
Available heap: 41408
Available heap: 41608
Available heap: 41768
Available heap: 41784
Available heap: 41000
Available heap: 40944
Available heap: 41008
Available heap: 40944
Available heap: 41000
Available heap: 40944
Available heap: 41000
Available heap: 40944
Available heap: 41592

Then I also tested with using the sent callback directly in `send(string[, function(sent)])` but the behavior was the same as in the first example.

devsaurus, 2016-05-19:
Try to add a print to the listen callback and check whether there are connections without actually receiving something:

```lua
srv:listen(80,function(conn)
    print("connected")
    conn:on("receive", function(c,request)
```

marcelstoer, 2016-05-19:
Thanks for the hint Arnim but that lead nowhere.

> Server created...
Available heap: 42328
=collectgarbage()
0
=node.heap()
44432
connected
Available heap: 43696
connected
Available heap: 43536
connected
Available heap: 42872
connected
Available heap: 42608
connected
Available heap: 42104
...
connected
Available heap: 40304
connected
Available heap: 41304
=collectgarbage()
0
=node.heap()
41440

Most of the requests are from `wget`, others from the Browser. There was a delay of several minutes between the last two requests because I stopped sending request (the available heap increased).

devyte, 2016-05-20:
@marcelstoer 
Shouldn't that server code look like this?
```
srv=net.createServer(net.TCP)
print("Server created...")
print("Available heap: "..node.heap())
srv:listen(80, function(conn)
        conn:on("receive", function(c,request)
                c:send("HTTP/1.1 200 OK\r\nContent-Type: text/plain\r\n\r\nLED STATE=1\r\n")
            end)
        conn:on("sent", function(c) 
                print("Available heap: "..node.heap())   
                c:close()
            end)
    end)
```
Notice that the callback settings aren't nested.

Also, I seem to remember somebody saying that connections shouldn't be closed from within one of its callbacks. I took that to mean that the final correct version should really look something like this:
```
srv=net.createServer(net.TCP)
print("Server created...")
print("Available heap: "..node.heap())
srv:listen(80, function(conn)
        conn:on("receive", function(c,request)
                c:send("HTTP/1.1 200 OK\r\nContent-Type: text/plain\r\n\r\nLED STATE=1\r\n")
            end)
        conn:on("sent", function(c) 
                print("Available heap: "..node.heap())   
                node.task.post(node.task.HIGH_PRIORITY, function() c:close() end)
            end)
    end)
```
I have no idea whether that's really true, but I do currently have a leak that I haven't been able to figure out in over 3 months. Posting a task is the only thing I have yet to try.

marcelstoer, 2016-05-20:
@devyte if the assumption your first code snippet is based upon were true then [our `socket:send` example in the docs](https://nodemcu.readthedocs.io/en/dev/en/modules/net/#example_6) would be wrong, no?

devsaurus, 2016-05-20:
@devyte's first example stabilizes the heap for me, see below.
I also remember a discussion regarding when to close the connection. Can't find it again, but it was mentioned not to close right after the `conn:send()` since this would cancel the queued send.

```
> Available heap: 43672
Available heap: 43464
Available heap: 43280
Available heap: 43104
Available heap: 42912
Available heap: 42736
Available heap: 42928
Available heap: 42752
Available heap: 42928
Available heap: 42752
Available heap: 42928
Available heap: 42760
Available heap: 42928
Available heap: 42752
Available heap: 42928
Available heap: 42752
Available heap: 42928
Available heap: 42760
Available heap: 42944
Available heap: 42752
Available heap: 42928
Available heap: 42752

---> long pause between requests <--

Available heap: 43664
Available heap: 43488
Available heap: 43296
Available heap: 43120
Available heap: 42928
Available heap: 42752
Available heap: 42928
Available heap: 42752
Available heap: 42936
Available heap: 42752
Available heap: 42928
Available heap: 42752
Available heap: 42928
```

marcelstoer, 2016-05-20:
Arnim I can confirm your findings, same here. To conclude we either have an issue in our code or we should really change the `socket:send` example:

```lua
srv = net.createServer(net.TCP)
srv:listen(80, function(conn)
  conn:on("receive", function(sck, req)
    local response = {}
    response[#response + 1] = "HTTP/1.1 200 OK\r\n"
    response[#response + 1] = "Content-Type: text/plain\r\n\r\n"
    response[#response + 1] = "LED STATE=1\r\n"
    local function send()
      if #response > 0
        then sck:send(table.remove(response, 1))
      else
        print("Available heap: "..node.heap())
        sck:close()
      end
    end
    sck:on("sent", send)
    send()
  end)
end)
```
yields

```
Available heap: 42456
Available heap: 41912
Available heap: 41368
Available heap: 40792
Available heap: 40256
Available heap: 40136
Available heap: 39600
Available heap: 39344
Available heap: 38832
Available heap: 38688
Available heap: 38176
Available heap: 38040
Available heap: 37520
Available heap: 37368
Available heap: 36856
Available heap: 36456
Available heap: 35960
=collectgarbage()
0
> =node.heap()
36672
> 
```
@TerryE can you help here?

TerryE, 2016-05-21:
@marcelstoer sorry, I thought that I'd replied to this last night by email, but this reply hasn't been posted.

`local function send()` is using two closed upvals: `sck` and `response` so these references will persist whilst a reference to send exists which it will do with the socket since it's in the on callback, and the current `net.socket` object doesn't clear these down when the `close()` is issued.  So
-   get rid of the `sck`upvalue by using the fact that it is also passed as a argument to the callback
-   make sure that the `response` table is dereference as an upval

```Lua
    local function send(sk)
      if #response > 0
        then sk:send(table.remove(response, 1))
      else
        print("Available heap: "..node.heap())
        sk:close()
        response = nil
      end
    end
```

marcelstoer, 2016-05-21:
Thank you Terry. What you're saying sounds logical. However, `sk` is nil when `send` is invoked.

TerryE, 2016-05-22:
Sorry I missed that the outer send should be `send(sck)`.

marcelstoer, 2016-05-22:
I updated our docs accordingly.

commit afd974c5 (2016-05-22): Correct socket:send example, fixes #1303
PR #1358 (2016-06-13): Documentation Change - Memory leak
Fixes #1303.

- [x] This PR is compliant with the [contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `https://nodemcu.readthedocs.io/en/master/en/modules/net/#netsocketsend`.

The documentation have changed. See #1303
commit 0d3e0bbb (2016-06-13): Documentation Change - Memory leak (#1358)

Fixes #1303
closed 2016-05-22 by marcelstoer