nodemcu/nodemcu-firmware #1304: mqtt:unsubscribe dumps

dvv, 2016-05-20:
```
c = mqtt.Client("foo", 120)
c:connect("192.168.1.4", 1883, 0)
-- NB: http://nodemcu.readthedocs.io/en/dev/en/modules/mqtt/#syntax_7
c:unsubscribe("foo")
-- PANIC: unprotected error in call to Lua API (attempt to call a nil value)
```

FrankX0, 2016-05-20:
You are unsubscribing from a topic you did not subscribe to first.
And you seem to mix the client name (foo) and a topic (foo).

dvv, 2016-05-21:
I described the sequence on pretty valid calls which led to sudden reset.

devsaurus, 2016-05-21:
Thanks for reporting @dvv.
It appears that the unsubscribe callback is triggered without being initialized earlier. `mud->cb_unsuback_ref` is only set when `c:unsubscribe()` [provides a callback function](https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/mqtt.c#L1283) - otherwise it remains uninitialized. I'll dig into a fix.

pjsg, 2016-05-21:
Fix merged to dev

closed 2016-05-21 by pjsg