nodemcu/nodemcu-firmware #1308: socket disconnection flushes itself?

dvv, 2016-05-22:
Hello!

Am trying to write code which reliably connects to a remote TCP socket and keeps trying to reconnect on failures:

```Lua
CONNECT = function(host, port)
  local fd = net.createConnection(net.TCP, 0)
  local connect = function()
    print("TRY", node.heap())
    fd:connect(port, host)
  end
  fd:on("receive", function(fd, s)
    print("IN", node.heap(), s)
  end)
  fd:on("sent", function(fd)
    print("OU", node.heap())
  end)
  fd:on("connection", function(fd)
    print("CON", node.heap())
  end)
  fd:on("reconnection", function(fd)
    print("REC", node.heap())
  end)
  -- disconnection restarts connection
  fd:on("disconnection", function(fd)
    print("DIS", node.heap())
    tmr.alarm(6, 1000, 0, connect)
  end)
  -- first attempt to connect
  tmr.alarm(6, 1000, 0, connect)
  return fd
end

CONNECT("host", 4444) -- NB: not open at host
```

I expected to see repeating `TRY` `DIS` but saw only `TRY` `DIS` `TRY`. So seems that the call to `fd:connect` chokes after `fd:on("disconnection", fn)` fired.

Wonder do we invalidate somehow `fd` in disconnection handler and is it intended?

TIA

devsaurus, 2016-05-22:
> Wonder do we invalidate somehow `fd` in disconnection handler

Apparently, yes. The code for "socket disconnected" frees memory that got initialized during `createConnection()`. It's a one-way road from "disconnected -> close -> delete" - the life cycle of a socket doesn't allow to re-connect, IMO.

> and is it intended?

Don't know. It's built this way, though it doesn't look symmetrical wrt API design. Maybe worth a request to support this use case with the upcoming net module reimplementation in #1080.

dvv, 2016-05-22:
I see, `socket` is not supposed to be reusable.
Thank you!

TerryE, 2016-05-22:
If you look at the code `reconnection` is really a bad description of what this is sockets either terminate with a normal close or an error close such as an abort or a timeout.  The way that the ESPCONN layer handles this is that the disconnection callback is fired on normal close and the reconnection callback otherwise.  I am holiday at the moment but I will finish sorting out the reimplementation when I get back home..

closed 2016-06-29 by marcelstoer