nodemcu/nodemcu-firmware #1311: HTTPS client handshake failed

2coolc1d1, 2016-05-24:
First, thank you for NodeMCU.  It is great!

There seems to be an issue with the HTTPS implementation (or a user error :) ).  When I run the following code with HTTP everything works as expected.  However, if I change the protocol to HTTPS it fails (output below).

I've grep'd through the code looking for the error strings (and sub strings) without any luck.  I'm guessing the error may be coming from openssl and is the reason I'm not seeing it in the NodeMCU code.

Dev branch compiled in the cloud with SSL and debugging on.  hardware: Sparkfun ESP8266 Thing

<pre>You successfully commissioned a NodeMCU custom build from the dev branch.
You selected the following 9 modules: cjson file gpio http net node tmr uart wifi.</pre>

Being new to Lua and the NodeMCU, I'm hoping this is a simple mistake on my part but get the feeling it is not.

I'm happy to help debug/fix if anyone can point me me in the right direction.

**CODE**
<pre>
local function printIPConf()
    print("IP: "..wifi.sta.getip())
    print("Mode: "..wifi.getmode())
    print("Channel: "..wifi.getchannel())
    print("The module MAC address is: " .. wifi.ap.getmac())
end

local function configDNS()
    net.dns.setdnsserver("8.8.8.8", 0)
    net.dns.setdnsserver("8.8.4.4", 1)
    print("DNS 1: "..net.dns.getdnsserver(0)) -- 8.8.8.8
    print("DNS 2: "..net.dns.getdnsserver(1)) -- 8.8.4.4
end

wifi.sta.eventMonStop("unreg all")

wifi.sta.eventMonReg(wifi.STA_IDLE, function() print("STATION_IDLE") end)
wifi.sta.eventMonReg(wifi.STA_CONNECTING, function() print("STATION_CONNECTING") end)
wifi.sta.eventMonReg(wifi.STA_WRONGPWD, function() print("STATION_WRONG_PASSWORD") end)
wifi.sta.eventMonReg(wifi.STA_APNOTFOUND, function() print("STATION_NO_AP_FOUND") end)
wifi.sta.eventMonReg(wifi.STA_FAIL, function() print("STATION_CONNECT_FAIL") end)

wifi.sta.eventMonReg(wifi.STA_GOTIP, function()
    printIPConf()
    configDNS()

    http.get("https://httpbin.org/ip", nil, function(code, data)
        if (code < 0) then
            print("HTTP request failed")
        else
            print(code, data)
        end
    end)
end)

wifi.sta.eventMonStart()

wifi.setmode(wifi.STATION)
wifi.sta.config("SSID_HERE","PASSWORD_HERE")
</pre>

**OUTPUT**
<pre>
SID_HERE 3
PASSWORD_HERE 10
 0
00:00:00:00:00:00
state: 5 -> 0 (0)
rm 0
> 
	wifi_event_monitor_handle_event_cb is called
STATION_CONNECTING
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 6
cnt 

connected with SSID_HERE, channel 1
dhcp client start...

	wifi_event_monitor_handle_event_cb is called
ip:192.168.2.84,mask:255.255.255.0,gw:192.168.2.1

	wifi_event_monitor_handle_event_cb is called
IP: 192.168.2.84
Mode: 1
Channel: 1
The module MAC address is: 5e:cf:7f:8b:20:9c
DNS 1: 8.8.8.8
DNS 2: 8.8.4.4
client handshake start.
client handshake failed
Error: SSL error 40
HTTP request failed
-1	nil
</pre>

neuhalje, 2016-05-28:
Have a look [here](https://github.com/esp8266/Arduino/issues/43), they have the same error (though its arduino, but maybe its a lead).

marcelstoer, 2016-05-30:
The client and the server don't have any common cipher suites. The Espressif SDK provides the following cipher suites

TLS_RSA_WITH_AES_128_CBC_SHA
TLS_RSA_WITH_AES_256_CBC_SHA
TLS_RSA_WITH_RC4_128_SHA
TLS_RSA_WITH_RC4_128_MD5

but none of them is supported by the server, see https://www.ssllabs.com/ssltest/analyze.html?d=httpbin.org.

->  some HTTPS connections will work, some won't

closed 2016-05-30 by marcelstoer

2coolc1d1, 2016-05-31:
Thank you for the info.