nodemcu/nodemcu-firmware #1315: cjson.decode accepts numbers?

dvv, 2016-05-26:
```Lua
print(cjson.decode(0x16))
-- 22
```

marcelstoer, 2016-05-26:
Our module is based on Lua CJSON. It's got [lots of unit tests](https://github.com/mpx/lua-cjson/blob/master/tests/test.lua) and two which show how they handle hex values (lines 223 & 242). Furthermore, [its `decode` documentation](http://www.kyne.com.au/~mark/software/lua-cjson-manual.html#_decode) says:

> By default, numbers incompatible with the JSON specification (infinity, NaN, hexadecimal) can be decoded. This default can be changed with cjson.decode_invalid_numbers.

dvv, 2016-05-26:
Thanks for quick reply!
However my point was that our `cjson.decode` accepts _numeric argument_ while it should bark at wrong type (imho) like it does if `cjson.decode({foo="bar"})` or `cjson.decode(nil)`.

devsaurus, 2016-05-26:
@dvv guess you'd need [cjson.decode_invalid_numbers()](http://www.kyne.com.au/%7Emark/software/lua-cjson-manual.html#decode_invalid_numbers) to revert cjson back to the strict behaviour. It's not exhibited by our module although the underlying C code is implemented.

Does anybody know why [this method is commented](https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/cjson.c#L1613)?

dvv, 2016-05-26:
https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/cjson.c#L1449
Oh my, they try their best to convert narg to string behind the scene ) Beautiful

marcelstoer, 2016-05-26:
We use Lua CJSON with its default behavior. I would hate to change the defaults.

If the request eventually is to expose `decode_invalid_numbers()` then you'll need pretty good arguments as for why to expose only this one and not the others.

dvv, 2016-05-26:
The issue was about stricter argument checking. I reread lualib docu and discivered luaL_checkstring tries to fetch string even argument is a number.
So I'm done with this.
No need to expose anything.

marcelstoer, 2016-05-26:
I created a follow-up #1316.

closed 2016-05-26 by marcelstoer