nodemcu/nodemcu-firmware #1316: Expose Lua CJSON config functions

marcelstoer, 2016-05-26:
Follow-up to #1315.

If there's no compelling reason why the various [Lua CJSON config functions](http://www.kyne.com.au/~mark/software/lua-cjson-manual.html#decode_invalid_numbers) should _not_ be exposed then we should do so.
Code pointer: https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/cjson.c#L1613

marcelstoer, 2016-05-26:
@funshine do you remember why you initially disabled those functions? It's been like that from day 1.

jmattsson, 2016-08-11:
A bit of a side-track, but I'd be more inclined to look at replacing cjson with e.g. JSMN (http://zserge.com/jsmn.html) which is far more suited for embedded systems. Error recovery with our current cjson is nigh-on impossible as there are abundant places where it fails to check error codes after memory allocation. Wrapping its memory allocation functions made things _better_, but not good - iirc it will still happily leak memory under certain circumstances.

marcelstoer, 2016-08-11:
Johny, I can't judge by the code but from what I read in the description this is _not_ a parser but a simple tokenizer (e.g. doesn't handle escapes/UTF-8).
There's a 4years old benchmark at http://lionet.livejournal.com/118853.html. Our [Lua cJSON](https://github.com/mpx/lua-cjson/) base has seen some recent updates we're not aware of (see #1411...). [cJSON](https://github.com/DaveGamble/cJSON) may be a real alternative but, as so often, Arduino libraries like https://github.com/bblanchon/ArduinoJson or https://github.com/squix78/json-streaming-parser may be better candidates, no?

Do you agree we should track this in a new issue?

jmattsson, 2016-08-11:
Sure! I've just been trying to catch up a bit on my github notifications...

marcelstoer, 2016-12-12:
That new issue is finally here: #1666 😜 

Depending on the outcome there _this_ issue will become obsolete.

closed 2017-02-18 by marcelstoer