nodemcu/nodemcu-firmware #1320: WPA Enterprise Encryption

Flole998, 2016-05-28:
As WPA2 Enterprise is now supported by the SDK it should be added to nodemcu. Issue #366 should be reopened.

Flole998, 2017-08-09:
To everyone waiting for those feature: Just don't use nodemcu, use C in Arduino, they have WPA Enterprise support (even with PEAP if I am not mistaken). Since I created the first issue which was closed I have switched to Arduino and it's working way better than nodemcu ever did. Still someone should invest time to implement this in nodemcu.

ds2k5, 2018-04-16:
Hi Flole998,
can you share sample Code please ?
Thanks

tolean, 2018-04-16:
I have developed branch for enterprise wifi, if you want I share it

b08Walls, 2018-09-06:
tolean please, share my friend :)

Flole998, 2019-09-01:
Instead of closing the issue someone should fix it....... While fixing means open the other issue again in this case. Until someone has done so, this one should stay open.

closed 2019-09-16 by stale[bot]