nodemcu/nodemcu-firmware #1325: add TLS cipherlist to documentation

hex2a, 2016-05-30:
please document the supported ciphers for TLS as many of them (especially RC4) are disabled in current distributions (eg arch linux with nginx).

http.get "https://www.howsmyssl.com/a/check" shows:
TLS 1.1

TLS_RSA_WITH_AES_128_CBC_SHA
TLS_RSA_WITH_AES_256_CBC_SHA
TLS_RSA_WITH_RC4_128_SHA
TLS_RSA_WITH_RC4_128_MD5

marcelstoer, 2016-05-30:
Well, it _is_ documented - just not where you might be expecting it: http://nodemcu.readthedocs.io/en/dev/en/modules/mqtt/#mqttclientconnect

The same information would have to be replicated to `net.md`, `http.md` and possibly others?

marcelstoer, 2016-06-01:
@pjsg you've been working on cert stuff not too long ago. How deep are those (very few) cipher suites burned into the SDK? Could the firmware add support for a few more on top of that?

pjsg, 2016-06-01:
I suspect that if we move away from the espconn stuff, then we have to implement our own TLS layer. the current layer is using axtls, and the list of ciphers supported appears to be the list that axtls supports. 

Unfortunately, I can't find a good list of ciphers to support to give good coverage. 

I'm assuming that this ticket is really about wanting to connect to some server that doesn't support any of these ciphers -- is there an example of this?

marcelstoer, 2016-06-01:
There's an example at #1311. Unfortunately(?) https://httpbin.org is a popular test server 😉 

This issue here is about *documenting* the supported suites. ATM we don't have a good place in the docs for such cross-cutting concerns. Of course we can link to http://axtls.sourceforge.net/specifications.htm but several modules are affected.

pjsg, 2016-06-01:
I just ran the server check on ssllabs: https://www.ssllabs.com/ssltest/analyze.html?d=httpbin.org

It looks as though the prolem is that axtls does not support DHE (aka EDH). I don't know how difficult that would be to add....

marcelstoer, 2016-06-01:
I just created a PR that fixes this temporarily by adding notes to the affected module documentation.

PR #1333 (2016-06-01): Add SSL/TLS remarks to docs
fixes #1325
closed 2016-06-01 by marcelstoer