nodemcu/nodemcu-firmware #1326: Socket send works as snippet in ESPlorer, not as part of .lua file

deviantp, 2016-05-30:
Before you assume this another async socket issue, here is the code:

`conn = net.createConnection(net.TCP, 0)  
conn:on("sent", function(sck,c) 
   print("Sent")
   sck:close() 
   end)

conn:on("connection", function(sck,c) 
   print("Connected..")
   sck:send("test") 
   end) 

 conn:connect(9090, "192.168.1.89")
 print("Send data.")`

When run as a snippet it connects OK and I receive the data on the server. When run as part of mcu-temp.lua , called by init.lua with 
`tmr.alarm(0,3000,0,function() dofile('mcu-temp.lua') end)` 
the socket connection event never gets triggered, so I don't see the "Connected.." message. Without resetting the board, I can then run the snippet (same code copy and pasted) and it runs fine. I can run it multiple times, and it works every time.

Here is the entirity of the mcu-temp.lua file
`--Print current WIFI configuration

ssid, password, bssid_set, bssid=wifi.sta.getconfig()
print("\nCurrent Station configuration:\nSSID : "..ssid
.."\nPassword  : "..password
.."\nBSSID_set  : "..bssid_set
.."\nBSSID: "..bssid.."\n")

tmr.delay(5000000)

print("Wifi-Status: " .. wifi.sta.status())

-- Get Temperature
 
print("Getting Temperature.")
status, temp, humid, temp_dec, humid_dec = dht.read(2)

print("DHT-Status : " .. status)
print("Temperature: " .. temp)
print("Humidity   : " .. humid)

--delim = "###"
--sendstring = dataset .. delim .. location .. delim .. temp .. delim .. humid .. "#"
--print("Sending data : " .. sendstring)
 
conn = net.createConnection(net.TCP, 0)  
conn:on("sent", function(sck,c) 
   print("Sent")
   sck:close() 
   end)

conn:on("connection", function(sck,c) 
   print("Connected..")
   sck:send("test") 
   end) 

 conn:connect(9090, "192.168.1.89")
 print("Send data.")`

Example output (ssid/pass redacted):
Current Station configuration:
SSID : xxxxxxxxx
Password  : xxxxxxxx
BSSID_set  : 0
BSSID: ff:ff:ff:ff:ff:ff

Wifi-Status: 5
Getting Temperature.
DHT-Status : 0
Temperature: 23
Humidity   : 30
Send data.

Immediately run snippet:

> conn = net.createConnection(net.TCP, 0) conn:on("sent", function(sck,c) print("Sent") sck:close() end)  conn:on("connection", function(sck,c) print("Connected..") sck:send("test") end)  conn:connect(9090, "192.168.1.89")  conn:send("test")
> Connected..
Sent

FrankX0, 2016-05-30:
I tried your code (but using port 80 and a web-server), and it runs fine.
I get the "Connected.." and "Sent"  messages.
Can you please try on the latest (dev) build? The one you are using is quite old.

deviantp, 2016-05-30:
Thanks for taking a look, I think it's actually related to how I'm checking for the connection state. I was polling in a loop with a tmr.delay. I'm closing the issue as it's fully working with a better tuned wifi initialization script.

closed 2016-05-30 by deviantp

marcelstoer, 2016-05-31:
For the record here's the corresponding SO question http://stackoverflow.com/q/37523837/131929 and our guideline http://nodemcu.readthedocs.io/en/dev/en/support/.