nodemcu/nodemcu-firmware #1328: Increase maximum filename length?

oyooyo, 2016-05-31:
nodemcu-firmware currently has a 31 character limit for filenames (see #1112), by simply using the default from SPIFFS, defined in SPIFFS_OBJ_NAME_LEN in file spiffs_config.h
I suggest increasing this limit, at least by a few characters, unless there's a good reason to keep it that low (I'm not aware of the implications/downsides this has on SPIFFS).

The reason why 31 characters don't seem sufficient to me is because I'm building a small LUA library for nodemcu where LUA modules are not identified by a typical human-readable name, but rather by a UUID, so my LUA code contains lines like:
`local Rgb_Led = require('3bb15b73-a83a-495e-a1b1-e4e35a196a4c')`
and my filesystem thus contains multiple files with names like "3bb15b73-a83a-495e-a1b1-e4e35a196a4c.lua", which requires 40 characters, so I need SPIFFS_OBJ_NAME_LEN to be >= 40+1=41.

marcelstoer, 2016-05-31:
Won't this be fixed once we merge #1226 (which also fixes #1164)?

oyooyo, 2016-05-31:
I don't think so. This appears to be the default config of the latest version of SPIFFS:
https://github.com/pellepl/spiffs/blob/master/src/default/spiffs_config.h
and the default for SPIFFS_OBJ_NAME_LEN is still defined as 32 there, so the latest version of SPIFFS still seems to have a 31 character maximum for filenames by default.

pjsg, 2016-06-01:
From looking at the current code, there are still problems with filenames of length 32 -- which should probably be fixed as part of the new spiffs upgrade.

It should be possible to increase the name length -- it seems that SPIFFS uses fixed size structures in the filesystem for storing object names.... So an increase to 64 (say) would consume more space for everybody.

oyooyo, 2016-06-01:
@pjsg: Thanks for your assessment. What kind of "more space" would it consume - just flash memory, or would increasing the maximum filename length automatically consume more precious RAM as well?

If for example an increase to 64 would only have an additional cost of 32 bytes of flash memory per actual file stored in SPIFFS, I would expect this to be a minor practical problem, since the average NodeMCU system probably just has a handful of files stored in SPIFFS anyway, and flash memory generally only seems to be limited on hardware with just 512kb flash.

If, however, allowing longer filenames would automatically cost more precious RAM as well, I would understand that it's too costly to have a higher value by default...

Hmmm, I guess I should just give it a practical try. I will do that and post my results later.

oyooyo, 2016-06-01:
Okay, so I did a test. I compiled three NodeMCU firmware images that were completely identical except for the SPIFFS_OBJ_NAME_LEN setting. I used three different settings for SPIFFS_OBJ_NAME_LEN: the current default of 32, 42 (which would be enough for my purposes) and 64 (as mentioned in this thread).

The compiled firmware images had the following sizes:
32: 503479 bytes
42: 503543 bytes
64: 503559 bytes
So increasing the maximum filename length from 32-1=31 to 64-1=63 characters resulted in a firmware image 80 bytes larger. I personally consider that neglectable.

Once each of the firmware images was flashed to my NodeMCU devkit, I uploaded a tiny program consisting of and actively using/accessing 11 individual LUA files. It dumps the current result of node.heap() and node.fsinfo() at various stages. For each of the three firmware versions, I ran the program three times by pressing the reset button and collecting the output.

Looking at the result, what's interesting is that **increasing the maximum allowed filename length seemed to not have any effect on both RAM/heap and SPIFFS storage space** - the values were completely identical. So I guess increasing the max. filename length only affects the structure of the SPIFFS metadata area, and maybe only decreases the number of files that can be stored in the filesystem or something like that?
The page on SPIFFS configuration only mentions the following about SPIFFS_OBJ_NAME_LEN:
`Maximum number of characters in a file name. Must absolutely not exceed the logical page size minus some bytes.`

marcelstoer, 2016-06-03:
> LUA modules are not identified by a typical human-readable name, but rather by a UUID

So, if the filenames aren't human readable anyway I'm sure you could hash them to fit into the 31-chars boundary?

oyooyo, 2016-06-03:
You're right of course. Actually, that was more or less the solution I used initially: I created a `require_uuid` function to be called with a UUID string, removed all non-hexadecimal characters, converted it to a 16 characters LUA binary string using encoder.fromHex(), then base64-encoded that binary string to a "normal" string of length 25 and then called `require` with that string.
So, for example, calling
`require_uuid('c865c505-4e81-488d-8a99-e1bdecd167e8')`
would actually result in calling
`require('yGXFBU6BSI2KmeG97NFn6A==')`
...which is compatible with the 31 characters filename limit. Unlike hashing, it also had the benefit that the transformation could also be reversed.

That is a functional workaround that solves the problem, yes. But... I don't know, when I realized I could simply increase the max. filename length limit with one short line in user_config.h, I felt like increasing the max. filename length limit is ultimately the simpler, better and less memory-consuming solution. No need for a workaround that relies on the existence of the encoder module, no additional function needed, and while they both aren't human-readable, somehow I still prefer the filename 'c865c505-4e81-488d-8a99-e1bdecd167e8.lua' over 'yGXFBU6BSI2KmeG97NFn6A==.lua'...

marcelstoer, 2016-11-19:
Philip, if you don't think we should change the file length then please close this. My opinion is in https://github.com/nodemcu/nodemcu-firmware/pull/1226#issuecomment-223771397.

closed 2017-02-18 by marcelstoer