nodemcu/nodemcu-firmware #1329: MQTT, QoS and Network Latency

NicolSpies, 2016-05-31:
-  When using MQTT on a network with typical latency of 10ms, one message is exchanged between the NodeMcu MQTT client and the MQTT Broker regardless of the QoS setting.
- When the same test is repeated on a slow network (3G) with a latency in the region of 140ms, multiple similar messages is exchanged (up to 19) between the client and broker regardless of the QoS setting.
- If I am understanding QoS correctly, a Qos = 2 should prevent this ensuring a message is received only once by the broker.
- If my reasoning is correct, does this point to a anomaly in the MQTT QoS operation ?

marcelstoer, 2016-05-31:
Which branch do you use? @pjsg fixed a number of MQTT issues in `dev` a few months back.

NicolSpies, 2016-05-31:
Latest dev on NodeMCU custom builds

NicolSpies, 2016-05-31:
It might be a case that the retry timeout (MQTT_SEND_TIMEOUT ?) is to short and needs to be increased to reduce network clogging with duplicate messages on slower networks

pjsg, 2016-06-01:
It appears that the retransmit timeout is 5 seconds -- which is plenty long enough for a 140ms RTT. Can you do a packet trace to see what is happening? I don't have an easy way of introducing a significant delay.... It also may depend on the way that you invoke the client API.

NicolSpies, 2016-06-01:
- **It is fixed**, the slower network triggered a subtle problem in the way I called the MQTT publish. I have messages in a table and a timer loop publishing and removing the message from the table when the callback function executes when the PUBACK is received.
 
- In the slower network the PUBACK takes longer, when the timer fires the message has not been removed and is sent again. This situation gets worse when QoS >0 is used for obvious reasons.
 
- **Solution**: slow the timer down to ensure that the PUBACK has been received and the message removed from the list.

marcelstoer, 2016-06-01:
Assuming I get this right then your timer isn't a real help, is it? Wouldn't you rather want to send message n once the PUBACK for n-1 has been received? If so then an approach like the one we documented for [`socket:send()`](http://nodemcu.readthedocs.io/en/dev/en/modules/net/#example_6) could be more useful?

closed 2016-06-01 by marcelstoer

NicolSpies, 2016-06-01:
You are correct. The process is asynchronous (detached), the messages for publication queued/added to the table and the publish loop servicing and removing messages as it is sent. The loop needs to change not being dependent on a timer but on the PUBACK as you stated. Making the timer longer just  desensitize the system to the problem but does not really fix it. Thanks for your suggestion.