nodemcu/nodemcu-firmware #1331: Wrong Lua interpreter output over socket

Serrgee, 2016-06-01:
I try telnet lua script with master/dev builds. It sends output strings without \n and does not send prompt "> " . This is `socat` debug session after connection and entering `print(node.heap())`:
>$ sudo socat -v - tcp4:192.168.5.232:2323
\< 2016/06/01 09:15:29.220369  **length=25** from=0 to=24
Welcome to NodeMcu world.Welcome to NodeMcu world.print(node.heap())
\> 2016/06/01 09:28:30.855140  length=19 from=0 to=18
print(node.heap())
< 2016/06/01 09:28:30.945686  length=5 from=25 to=29
3946439464

The same session with firmware 0.9.6 looks like this:
>$ sudo socat -v - tcp4:192.168.5.218:2323
< 2016/06/01 09:41:17.162748  **length=26** from=0 to=25
Welcome to NodeMcu world.
Welcome to NodeMcu world.
print(node.heap())
\> 2016/06/01 09:41:36.864300  length=19 from=0 to=18
print(node.heap())
\< 2016/06/01 09:41:37.053738  length=5 from=26 to=30
2984029840< 2016/06/01 09:41:37.058160  length=3 from=31 to=33
\>

devsaurus, 2016-06-01:
I assume that you use the updated Lua example from #909? The newer SDKs didn't play well with the old examples.

Please post your scripts in case the problem persists.

joep, 2016-07-05:
This appears to be related to setting the cpu freq to 160 mhz. When I added the command to do so in init.lua, the prompt disappears when I telnet in. If I set it back to 80, it reappears.

marcelstoer, 2016-07-11:
No more feedback -> closing.

closed 2016-07-11 by marcelstoer