nodemcu/nodemcu-firmware #1334: file system corruption?

joysfera, 2016-06-01:
I think I'm experiencing a filesystem corruption in 
```
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 08aff66c29f5a29279427e584b2e3bd3a6ca5719
	SSL: false
	modules: file,gpio,http,net,node,ow,tmr,uart,wifi
 build 	built on: 2016-05-28 20:25
 powered by Lua 5.1.4 on SDK 1.5.1(e67da894)
```

My code implements a primitive web server: in on("receive") handler it decodes the client reply and saves a three configuration files. When this happens twice then a compiled lua file (that is normally uploaded as the first file ever on a cleanly formatted filesystem) gets corrupted. At first its file length becomes zero but later also dofile("ds18b20.lc") fails so it is indeed corrupted, IMHO. And it also seems that one of the newly created files ("names.cfg") disappeared completely from the filesystem after the second client reply:

initial file list:
```
main.lc	5148
ds18b20.lc	1240
s/style.css	676
s/config.html	4973
init.lua	732
wificonfig.lc	4040
s/index.html	970
temperwebserver.lc	7840
```

after first client reply:
```
init.lua	732
export.cfg	2
s/style.css	676
s/index.html	970
main.lc	5148
ds18b20.lc	1240
s/config.html	4973
rules.cfg	23
wificonfig.lc	4040
names.cfg	0
temperwebserver.lc	7840
````

after second client reply:
```
init.lua	732
export.cfg	2
s/style.css	676
s/index.html	970
main.lc	5148
ds18b20.lc	0
s/config.html	4973
wificonfig.lc	4040
rules.cfg	23
temperwebserver.lc	7840
```
As you can see, the "ds18b20.lc" has zero length now. And names.cfg is missing.

I wonder if I did something wrong. Is it OK to create a file at runtime? Or three files subsequently? I doubt it's normal that an existing file's length becomes zero or that another file disappears completely. What's up?

This is with ESP8266-07 and the filesystem info is as follows:
```
File system info:
Total : 534881 Bytes
Used : 29618 Bytes
Remain: 505263 Bytes
```

oyooyo, 2016-06-02:
I cannot really help, but I believe to have experienced the same or at least a closely related problem yesterday. My code was a little test program that wrote a few test files at runtime, always using the same filenames. When I ran this code multiple times, a completely different LUA file in the filesystem suddenly became corrupted. Just like in your case, it suddenly had a file size of 0 and couldn't be executed anymore.

In my case, I made an interesting observation: When I manually deleted those test files that I had written at runtime, the seemingly corrupted LUA file suddenly had it's original size and worked fine again...

Since it was just a short test program and I already got the results I was interested in, I didn't give any more attention to it, so I too don't know if it's an actual bug or I simply did something wrong.
I suspected that the problem might be related to the fact that those test files that I wrote at runtime were opened using 'w' mode:
`file.open(..., 'w')`
instead of 'w+' mode:
`file.open(..., 'w+')`
because when I wrote the test program and read the documentation for file.open, I was quite confused which of the two write modes I should choose and simply picked 'w' mode. _(I really think the documentation for the mode parameter of file.open should be extended to clarify the difference between the 'r','w','a' and the 'r+','w+' and 'a+' modes, it's not obvious!)_

But this is just a guess of course and probably doesn't solve the problem - but as long as no one else has a solution, you might still give it a try.

joysfera, 2016-06-02:
I have tried removing the three *.cfg files that were created at runtime (with file.open(..,"w")) but it didn't help to recover the ds18b20.lc. What's worse - the file seems to become unremovable:
```
> l = file.list(); for k,v in pairs(l) do  print("name:"..k..", size:"..v) end
name:main.lc, size:5148
name:ds18b20.lc, size:0
name:s/style.css, size:676
name:s/config.html, size:4973
name:init.lua, size:732
name:wificonfig.lc, size:4040
name:s/index.html, size:970
name:temperwebserver.lc, size:7752

> file.remove("ds18b20.lc")

> l = file.list(); for k,v in pairs(l) do  print("name:"..k..", size:"..v) end
name:main.lc, size:5148
name:ds18b20.lc, size:0
name:s/style.css, size:676
name:s/config.html, size:4973
name:init.lua, size:732
name:wificonfig.lc, size:4040
name:s/index.html, size:970
name:temperwebserver.lc, size:7752
```
You can see the ds18b20.lc is still there even though I removed it.

joysfera, 2016-06-02:
I have just replaced all "w" in my open() with "w+" but it hasn't helped a bit.

joysfera, 2016-06-02:
**Maybe** I found something: I haven't tested it deeply enough yet but I **think** the corrupted/masked file is caused by **writing an empty (zero size) file**. Going to test this theory at a newer firmware and with enabled all the features of my program I have disabled in last hours. Will report back. Keeping my fingers crossed and glueing back my pulled out hair.

devsaurus, 2016-06-02:
There's #1226 pending which fixes a couple of reported issues with our current spiffs integration. Your problem might be related even though the symptoms are kind of unique IMO.
It might take another week or two until the fixes are available in dev. Thus, if you're brave enough then give Philip's branch a try.

pjsg, 2016-06-03:
There are a number of issues with the current SPIFFS code. Part of the issues are due to the fact that the code tends to assume that the area of flash used for the file system is valid -- and this is not always the case. In these sort of cases, make sure that you format the file system.... 

If you can produce a test case that I can try on the new code, I would love to try it....

oyooyo, 2016-06-03:
> file.open("names.cfg", "w")
> file.flush()
> file.close()

Just an idea, but could this be somehow related to pellepl/spiffs#74? The symptom is different, but what your minimal code sample is doing reminds me of what pellepl claimed to be causing pellepl/spiffs#74 :

> It was a sneaky one, **will only happen when opening a only-created-never-written file** with SPIFFS_TRUNC flag.

pjsg, 2016-06-03:
I tried this test case, and it worked as you would hope with the new version of spiffs.

I also fixed the merge conflict with dev.....

devsaurus, 2016-06-09:
@joysfera can you confirm that your issue is fixed now in `dev`?

marcelstoer, 2016-08-10:
No feedback from OP in two months -> closing.

NicolSpies, 2016-08-10:
@marcelstoer I, experienced this yesterday again using the latest dev version with the following code,
```
    file.open(JsonFile..".json","w+")
    file.writeline(cjson.encode(table))
    file.flush()
    file.close()
```
The effect is that the file is saved as a empty file with 0 byte size.
My non-elegant work around in is to restore a backup of the file that corrupted.
I have experienced this 3 times in the last 12 months, I will keep tracking and reporting this problem on occurrence. I suggest we keep this open for a while now that I discovered it has been an issue.

pjsg, 2016-08-10:
Do you keep overwriting the same file? Also, how big is the json blob that is written? Is the filesystem nearly empty or nearly full? How big is the filesystem?

marcelstoer, 2016-08-10:
> I suggest we keep this open for a while now that I discovered it has been an issue.

Sure, as long as we have confirmation that it isn't fixed.

NicolSpies, 2016-08-10:
Yes same file is being overwritten on a regular basis. 4Mb file system, using less than 100kb. json blob kept below 1kb

pjsg, 2016-08-11:
I took a look at the cjson.encode functionality and it seems that it is a bit short on error checking. I was wondering if sometimes it fails to encode the string and returns an empty string. However, the error return would probably cause a fault.

TerryE, 2019-06-08:
I am going to close this one.  We've upgraded SPIFFS a couple of times since this report. No doubt someone can reopen it or raise a new issue of this reoccurs.

closed 2019-06-08 by TerryE