nodemcu/nodemcu-firmware #1338: sending messages from both side, one esp to another

dkumariet, 2016-06-04:
i am not able to send message from server(AP) to client , using hyper terminal, simply server replied to client, when any message received from client 

>send("hi") //send message "Hi" from client using hyper-terminal 
>received: Hi, I read You //received message from server

but not sending message from server to client just like above 

 this is my server(AP) code
### server
````
print("configuring ap")

wifi.setmode(wifi.STATIONAP)
cfg={}
cfg.ssid="WS"
cfg.pwd="12345678"
wifi.ap.config(cfg)

print("config TCP server")
server = net.createServer(net.TCP,28800)
server:listen(80, function(conn)
     conn:on("receive",function(conn,data)
          print(data)
          message = "Hi, I read You."
        conn:send(message); message = nil

     end)
     conn:on("sent",function(conn)
    
          collectgarbage()
     end)
          conn:on("connection",function(conn)
          print("Connected")
     end)
````

dnc40085, 2016-06-04:
please see http://nodemcu.readthedocs.io/en/master/en/support/

dkumariet, 2016-06-04:
yes i have seen, 
but not found any information about my problem,
i want to send data from server(with AP mode) to client.
for EXM:- hyper terminal to ESP(AP) to ESP(STA),to hyper terminal
this is working Hyper terminal to ESP(STA) to ESP(AP) to Hyper terminal

dkumariet, 2016-06-04:
@dnc40085 yes i have seen, 
but not found any information about my problem,
i want to send data from server(with AP mode) to client.
for EXM:- hyper terminal to ESP(AP) to ESP(STA),to hyper terminal
this is working Hyper terminal to ESP(STA) to ESP(AP) to Hyper terminal

dnc40085, 2016-06-04:
from  http://nodemcu.readthedocs.io/en/master/en/support/
>The issues list on GitHub is not the right place to ask for help. Use it to report bugs and to place feature requests. Questions like "how do I ..." or "I can't get this to work ..." should be directed to [stackoverflow.com](http://stackoverflow.com/) or the NodeMCU section on [esp8266.com](http://www.esp8266.com/viewforum.php?f=17)

As stated above. This is not the right place to ask this type of question.

dkumariet, 2016-06-04:
@dnc40085 ok, thanks

closed 2016-06-04 by dkumariet