nodemcu/nodemcu-firmware #1340: Setting hostname in init.lua not supported?

marcelstoer, 2016-06-04:
An issue reported on bbs.nodemcu.com caught my attention and I'm not sure if it's related to #1259 that @dnc40085 fixed.

Are there any (new) constraints as for at what point the hostname can be set in station mode? None are currently documented. The following code in `init.lua` does NOT set the hostname to "mynode":

```lua
wifi.setmode(wifi.STATION);
wifi.sta.config("SSID","password")
wifi.sta.sethostname("mynode")

tmr.alarm(1,1000,1,function()
  if wifi.sta.getip()==nil then
     print("IP unavailable, waiting")
  else
     tmr.stop(1)
     print("Config done, IP is "..wifi.sta.getip())
     print("hostname: "..wifi.sta.gethostname())
  end
end)
```
It prints
```
IP unavailable, waiting
IP unavailable, waiting
IP unavailable, waiting
IP unavailable, waiting
Config done, IP is 192.168.0.56
hostname: NODE-B62BF
```
Once the device has booted I can run 

```lua
wifi.sta.sethostname("mynode")
print("hostname: "..wifi.sta.gethostname()) -- prints 'mynode'
```
but the home router's DHCP table still lists 192.168.0.56 -> NODE-B62BF of course. Moving the `sethostname` command to the first line has no effect btw.

devsaurus, 2016-06-04:
See my comment in #1259, try with autoconnect disabled.

marcelstoer, 2016-06-04:
Thanks, I admit that I only briefly re-scanned that issue before I created this one. Anyway, temporarily disabling auto-connect seems to make no difference:

```lua
wifi.sta.sethostname("mynode")
wifi.setmode(wifi.STATION);
wifi.sta.config("SSID","password", 0)
wifi.sta.connect()

tmr.alarm(1,1000,1,function()
  if wifi.sta.getip()==nil then
     print("IP unavailable, waiting")
  else
     tmr.stop(1)
     print("Config done, IP is "..wifi.sta.getip())
     print("hostname: "..wifi.sta.gethostname())
     wifi.sta.autoconnect(1)
  end
end)
```

Still uses hostname NODE-B62BF.

dnc40085, 2016-06-04:
I will look in to this and try and figure it out

dnc40085, 2016-06-04:
I tried the following code on: a current local dev build, a current master cloud build, and a current dev cloud build:
```lua
wifi.setmode(wifi.STATION);
wifi.sta.config("wifi","pass")
wifi.sta.sethostname("mynode")

tmr.alarm(1,1000,1,function()
  if wifi.sta.getip()==nil then
     print("IP unavailable, waiting")
  else
     tmr.stop(1)
     print("Config done, IP is "..wifi.sta.getip())
     print("hostname: "..wifi.sta.gethostname())
  end
end)
```
I had no issues at all, every time I got the same response (current master cloud build displayed):
```lua
NodeMCU custom build by frightanic.com
	branch: master
	commit: cdaf6344457ae427d8c06ac28a645047f9e0f588
	SSL: false
	modules: file,gpio,net,node,tmr,uart,wifi
 build 	built on: 2016-06-04 22:17
 powered by Lua 5.1.4 on SDK 1.5.1(e67da894)
> IP unavailable, waiting
IP unavailable, waiting
IP unavailable, waiting
IP unavailable, waiting
IP unavailable, waiting
Config done, IP is 10.0.0.5
hostname: mynode
```

dnc40085, 2016-06-04:
>Are there any (new) constraints as for at what point the hostname can be set in station mode? None are currently documented.

As far as I know there are no constraints, aside from my just discovered requirement that the hostname can only be set in station mode, meaning that if the ESP8266 is started in SoftAP mode then switched to Station Mode, the default hostname will remain "ESP_xxxxxx".

I guess #1317 didn't completely fix the problem... I'll have a fix ready for this soon.

@marcelstoer is the master branch stuck with this bug or should I submit the fix to master as well?

marcelstoer, 2016-06-04:
We don't land PRs on master. I don't remember a case where the a bug was critical enough to annul this rule. With our new [release guidelines](https://github.com/nodemcu/nodemcu-firmware#releases) it shouldn't take as long as last time until we snap dev back to master next time 😜 

Btw, I reproduced your findings with current dev & master (nearly identical because of yesterday's snap). I managed to trace the previous (locally built) binary back to May 24th/25th - which is 1/2 days before #1317 was merged, grrr.

dnc40085, 2016-06-04:
I have the fix ready for my newly discovered bug and I have submitted a PR to the dev branch

marcelstoer, 2016-06-05:
> Only reason I was wondering if the fix is going make it to master is because of concerns that new issues might start popping up informing us of the bug.

That does happen from time to time, indeed. We just tell people to use `dev` instead.

PR #1341 (2016-06-05): Fixed bug where default hostname is not set on boot when in SoftAP mode
Fixes #1340.

- [x] This PR is compliant with the [contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) (if not, please describe why).
- [X] I have thoroughly tested my contribution.

This PR is to fix a bug where default hostname is not set if ESP8266 is booted up in SoftAP mode.
closed 2016-06-05 by marcelstoer