nodemcu/nodemcu-firmware #1342: Extend ws2812 module with shift feature

skycoders, 2016-06-05:
### Missing feature
The ws2812 module was recently updated and the buffer concept was introduced. It also added a fade() method to fade out the buffer. As another extension I have implemented a shift method, which allows you to circulate the buffer forward (positive parameter) and backwards (negative parameter). This way you can quickly achieve animations, or move the buffer and overwrite the new first/last pixel afterwards.   
I have further extended the module to support fadeIn and fadeOut (fade method kept for backwards compatibility). I can provide a pull request for this, if this is seen as useful.

### Justification
fadeIn completes the already provided fadeOut. It will of course only work if the initial set of bytes allows to scale up, but nice animations will still be possible. 
The shift method is very useful to perform this task on e.g. LED circular modules or to do any other LED chasing features.

### Workarounds
This can be done in LUA, but it is runtime critical and not as memory efficient. For fast animations, it is better to have this in native code.

marcelstoer, 2016-06-05:
> I can provide a pull request for this, if this is seen as useful.

👍

devsaurus, 2016-06-05:
Also a go from my side. Please check pending #1310 for any interdependence.

skycoders, 2016-06-06:
As I have only changed buffer behavior but not the actual write command, this should not cause any issues with #1310

skycoders, 2016-06-06:
as this is my first time ever, I hope this pull request is OK :)

PR #1343 (2016-06-10): added fadeIn, faceOut and shift to ws2812 module
Fixes #1342.

- [x] This PR is compliant with the [contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

This PR adds a shift (forward and backward) to the ws2812 buffer. It simply circulates the data like in a ring. It also adds a fadeIn (and for consistency fadeOut) method. The fade method has been kept for backward compatibility.

Committers supportin [... cut]
closed 2016-06-10 by marcelstoer