nodemcu/nodemcu-firmware #1345: Crash on mqtt.close

helmutca, 2016-06-07:
### Expected behavior
don't crash at an mqtt.close within a loop

### Actual behavior
Output
============
Message send

ets Jan 8 2013, rst cause:2, boot mode:(3,6) 

load 0x40100000, len 29664, room 16
tail 0
chksum 0x24
load 0x3ffe8000, len 2184, room 8
tail 0
chksum 0xa3
load 0x3ffe8888, len 8, room 8
tail 0
chksum 0x08
csum 0x08
don't use rtc mem data

### Test code
Content of t1.lua
====================
function sent()
 status,temp,humi = dht.read("4")
 m:publish("sensor/temperature/1",temp,0,0)
 print("Message send")
end

Content of t2.lua
===================
m = mqtt.Client("1", 0, "xxx", "xxxxxxxxx")
m:connect("xxx.xxx.xxx.xx",1883,0,0,sent(),function(client, reason) print("failed reason: "..reason) end)
m:close()

>dofile("t1.lua")
>dofile("t2.lua")

crash happens during dofile("t2.lua"), during the m:close(). 

### NodeMCU version
Tested with stable and dev branch
NodeMCU custom build by frightanic.com
branch: dev
modules: dht,file,gpio,mqtt,net,node,rtcfifo,rtcmem,rtctime,sntp,tmr,uart,wifi, build on 2016-06-06 19:41
powered by Lua 5.1.4 on SDK 1.5.1(e67da894)

### Hardware
NodeMCU ESP-12, with dht22

oyooyo, 2016-06-07:
I don't know if this is the actual problem, but I believe you are this wrong anyway. From my understanding, mqtt.connect is an asynchronous function that will return more or less instantly, when the connection has not yet been established - so your m:close() in the next line call will try to close the connection before it is even established.

Try this instead:
```
m = mqtt.Client("1", 0, "xxx", "xxxxxxxxx")
m:connect("xxx.xxx.xxx.xx",1883,0,0,function()
  sent()
  m:close()
end, function(client, reason) print("failed reason: "..reason) end)
```
and see if it makes any difference.

helmutca, 2016-06-07:
Thanks for the feedback. With the provide code, it doesn't crash anymore. But I looks like, the m:close() doesn't work. In case I check the logfile of the mqtt server it says: 
""Socket error on client 1, disconnecting", so the result of the close is a socket error and not the close of the connection.

In case I do it without the close, the Serverlog said:
>>Client 1 already connected, closing old connection.
>>Client 1 disconnected.
>>New client connected from xxx.xxx.xxx.xx

oyooyo, 2016-06-07:
> But I looks like, the m:close() doesn't work. In case I check the logfile of the mqtt server it says:
> ""Socket error on client 1, disconnecting", so the result of the close is a socket error and not the close of the connection.

Sounds plausible. But I just realized there is another eventual problem in the code sample I suggested: Just like mqtt.client:connect(), mqtt.client:publish() accepts a callback function that is called when the message has actually been published. When doing
```
sent()
m:close()
```
in your code, you might eventually (I'm not really sure about this) close the connection before the message has actually being published. That might also explain the socket error in the logfile.
Give this a try, just to rule that eventual problem out:
```
m = mqtt.Client("1", 0, "xxx", "xxxxxxxxx")
m:connect("xxx.xxx.xxx.xx",1883,0,0,function()
  print("Connected to MQTT broker")
  local status,temp,humi = dht.read("4")
  local message_published = m:publish("sensor/temperature/1",temp,0,0, function()
    print("Message sent")
    local connection_closed = m:close()
    print("Connection successfully closed:", connection_closed)
  end)
  print("Message successfully published:", message_published)
end, function(client, reason) print("failed reason: "..reason) end)
```

marcelstoer, 2016-06-08:
The docs at http://nodemcu.readthedocs.io/en/dev/en/modules/mqtt/#mqttclient give you a reasonably good idea about how to use the module. Yes, everything is asynchronous. If you want close after publishing you should do so in the `publish` callback.

helmutca, 2016-06-08:
The last code solved it, thanks for explaining.

dvv, 2016-06-08:
Bug is that `mqtt` instance is allowed to be in non-determinate state -- how does one know whether it is safe to call anything? I'd rather make `mqtt:close()` a no-op if it's unsafe to call it.

oyooyo, 2016-06-08:
> Yes, everything is asynchronous. If you want close after publishing you should do so in the publish callback.

There's something I didn't quite understand when I looked at `publish()` btw: The documentation states that the publish() function "Returns true on success, false otherwise", which rather sounds like it's a synchronous function. Why is that, how can the function return the success status if it's asynchronous?

pjsg, 2016-06-08:
This is clearly a bug in the implementation of the mqtt module (it should never crash). I'm not certain as to what "ought to happen". 

I suspect that the m:close() ought to cancel any pending operations and then close the socket. Th other option is that m:close() is also asynchronous and just initiates a close process which completes at some indefinite time in the future. I think that the MQTT model probably leans towards the latter approach due to the QoS guarantees that it provides.

Opinions?

marcelstoer, 2016-06-08:
> This is clearly a bug in the implementation of the mqtt module

It isn't so clear to me but I agree that it's unclear what ought to happen.

> `publish()` function "Returns true on success, false otherwise"

That's odd indeed.

> `m:close()` ought to cancel any pending operations and then close the socket

That would lead to people asking why a sequence of `publish(), close()` would lead to messages being discarded.

I think one needs to look at the big picture i.e. how do you typically interact with the broker. Isn't it that you create the client and open a connection once the device boots? Then you keep publishing (IRQ- or timer-driven) for as long as the device is alive rather than connect-publish-close for every message? How should re-connect work if `close()` were asynchronous?

pjsg, 2016-06-08:
I think that after m:close() is called, no further API operations (on that client) should be permitted. You have to create a new Client object to make another connection. In particular, m:close() would prevent any auto reconnect. 

I suppose that I am thinking of m:close() as m:destroy()

pjsg, 2016-06-08:
With regard to `publish` returning true or false. There are conditions under which it fails (though it isn't clear to me whether it should actually be throwing errors in those cases). The `true` return means that it has enqueued the message to be sent, and it will be sent according to the QoS. The ack callback will be called at some future time.

dvv, 2016-06-08:
I believe https://www.npmjs.com/package/mqtt#example is a no-surprise adopted. May be author after it?

marcelstoer, 2016-06-08:
> The true return means that it has enqueued the message to be sent

That's what I thought but we need to make that a little clearer in the docs (same for other MQTT functions  that return `true`/`false`).

> it should actually be throwing errors in those cases

I'd prefer that.

> I suppose that I am thinking of m:close() as m:destroy()

👍 That would make the API a lot less ambiguous.

closed 2019-06-21 by stale[bot]