nodemcu/nodemcu-firmware #1346: EUS fails with latest dev build

FrankX0, 2016-06-08:
### Description
With the latest dev build enduser_setup.start() fails with an out of memory (if I'm not mistaken). Only the test code is loaded. Operation is successful with the current master build.
### Expected behavior
```Lua
> dofile("enduser.lua")
> Connected to wifi as:192.168.1.114
```
### Actual behavior
```Lua
> dofile("enduser.lua")
E:M -9904
```
### Test code
```Lua
enduser_setup.start(
  function()
    print("Connected to wifi as:" .. wifi.sta.getip())
  end,
  function(err, str)
    print("enduser_setup: Err #" .. err .. ": " .. str)
  end
)
```
### NodeMCU version
Build which fails:
```Lua
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 95d48ada69d741146703c89282d058074ebd206b
	SSL: false
	modules: enduser_setup,file,mdns,net,node,rtctime,sntp,tmr,uart,wifi
```
Build which works correctly:
```Lua
NodeMCU custom build by frightanic.com
	branch: master
	commit: cdaf6344457ae427d8c06ac28a645047f9e0f588
	SSL: false
	modules: enduser_setup,file,mdns,net,node,rtctime,sntp,tmr,uart,wifi
```

marcelstoer, 2016-06-09:
I believe this is caused by #1348.

Godzil, 2016-06-09:
I confirm, creating the enduser_setup.html file will make the enduser_setup.start() to work...

closed 2016-06-09 by marcelstoer