nodemcu/nodemcu-firmware #1347: Search in the docs broken

marcelstoer, 2016-06-09:
Due to https://github.com/rtfd/readthedocs.org/issues/1088 the search in our docs on RTD has been broken from the beginning. Yesterday I finally implemented an ugly work-around for `dev`: 7dd89dd15ef993c9740ba4d4361e6be8edd1284b.

If you now go to http://nodemcu.readthedocs.io/en/dev/ and search for 'gyroscope' you'll get a more or less expected result. If you do the same on http://nodemcu.readthedocs.io/en/master/ it'll fail.

Fellow committers, do you think this fix is important enough to be merged back to `master` right away i.e. before the next snap in seven weeks?

devsaurus, 2016-06-09:
This is an inconvenience for the user IMO. But neither wrong nor outdated Information which would break user code.
Thus I'd consider it as not critical & not urgent. Nonethless, thanks a lot for spotting and fixing it. This web coding is a miracle to me :wink:

marcelstoer, 2016-06-09:
> Thus I'd consider it as not critical & not urgent.

I agree but wanted to verify my assessment with the group.

marcelstoer, 2016-06-11:
I interpret silence as agreement with Armin (and myself) -> closing.

closed 2016-06-11 by marcelstoer