nodemcu/nodemcu-firmware #1348: problem in enduser_setup.c

apwiede, 2016-06-09:
### Expected behavior
In the following code in app/modules/enduser_setup.c the if clause should be entered, if the file does not exist:

    static int enduser_setup_http_load_payload(void)
    {
      ENDUSER_SETUP_DEBUG("enduser_setup_http_load_payload");
  
      int f = fs_open(http_html_filename, fs_mode2flag("r"));
      int err = fs_seek(f, 0, FS_SEEK_END);
      int file_len = (int) fs_tell(f);
      int err2 = fs_seek(f, 0, FS_SEEK_SET);
  
      const char cl_hdr[] = "Content-length:%5d\r\n\r\n";
      const size_t cl_len = LITLEN(cl_hdr) + 3; /* room to expand %4d */
  
      if (f == 0 || err == -1 || err2 == -1)
      {
         ....
       }

### Actual behavior
the if clause is not entered when the file does not exist.

### Reason:
this is because the macros in spiffs module have been changed:

old:

#define SPIFFS_API_CHECK_RES_UNLOCK(fs, res) \
if ((res) < SPIFFS_OK) { \
  (fs)->err_code = (res); \
  SPIFFS_UNLOCK(fs); \
  return -1; \
}

new:

#define SPIFFS_API_CHECK_RES_UNLOCK(fs, res) \
if ((res) < SPIFFS_OK) { \
  (fs)->err_code = (res); \
  SPIFFS_UNLOCK(fs); \
  return (res); \
}

as a result the fs_open and fs_seek calls - which use these macros in spiffs code - return different values like -10002

### possible FIX
I also think in the code above it does not make sense to call fs_seek and fs_tell, if the file does not exist. 
So I think the code could look like so:
 
    static int enduser_setup_http_load_payload(void)
    {
      ENDUSER_SETUP_DEBUG("enduser_setup_http_load_payload");
  
      int err = (FS_OPEN_OK -1);
      int err2 = (FS_OPEN_OK -1);
      int file_len = 0;
      int f = fs_open(http_html_filename, fs_mode2flag("r"));
      if (f >= FS_OPEN_OK) {
          err = fs_seek(f, 0, FS_SEEK_END);
          file_len = (int) fs_tell(f);
          err2 = fs_seek(f, 0, FS_SEEK_SET);
      }
  
      const char cl_hdr[] = "Content-length:%5d\r\n\r\n";
      const size_t cl_len = LITLEN(cl_hdr) + 3; /* room to expand %4d */
  
      if (f < FS_OPEN_OK || err < FS_OPEN_OK || err2 < FS_OPEN_OK)
      {
         ....
       }

### Test code
```Lua
    enduser_setup.start() returns E: M 904
 ```
### NodeMCU version
dev branch latest version

Cheers, 
Arnulf

apwiede, 2016-06-09:
I am also missing a fs_close call after the fs_read at the end of the same function. That might possible force memory leakage!

Godzil, 2016-06-09:
That would explain why I'm geting "E:M -9904" when trying to use enduser_setup.start()...

cmisztur, 2016-06-26:
same experience here

marcelstoer, 2016-06-27:
@robertfoss @pjsg could one of you verify the proposed fix and prepare a PR?

closed 2016-06-28 by marcelstoer