nodemcu/nodemcu-firmware #1349: MQTT Client never freed (memory leak)

djphoenix, 2016-06-09:
### Expected behavior
After `failure` callback or `:close()` method and nullified pointer, all memory that used by client is freed (heap before creation == heap after destroy)

### Actual behavior
Memory leak

### Test code
mqtt.lua
```lua
local client = mqtt.Client( 'Test client', 120, nil, nil, 1 )
client:on('connect', function(client)
    print('CONN')
    client:subscribe("esp/mqtt/test", 0)
    client:publish("esp/mqtt/test", "test", 0, 0)
end)
client:on('message', function(client, topic, message)
    print("RECV", topic, message)
    client:close()
end)
return client
```
mqtt-unsecure.lua
```lua
dofile('mqtt.lua'):connect( "85.119.83.194", 1883,
    0, 0, nil,
    function(client, err)
        print("CONN FAIL", err)
    end
)
```
mqtt-secure.lua
```lua
dofile('mqtt.lua'):connect( "85.119.83.194", 8883,
    1, 0, nil,
    function(client, err)
        print("CONN FAIL", err)
    end
)
```

example output (disconnected):
```
> =node.heap()
43640
> dofile("mqtt-unsecure.lua")
CONN FAIL	-5
> = collectgarbage(), node.heap()
0	42696
> dofile("mqtt-secure.lua")
CONN FAIL	-5
> = collectgarbage(), node.heap()
0	41504
... after some time ...
> = collectgarbage(), node.heap()
0	41704
```
example output (wifi connection):
```
= collectgarbage(), node.heap()
0	43520
> dofile("mqtt-secure.lua")
CONN
RECV	esp/mqtt/test	test
> = collectgarbage(), node.heap()
0	43480
> dofile("mqtt-unsecure.lua")
CONN
RECV	esp/mqtt/test	test
> = collectgarbage(), node.heap()
0	43264
```
### NodeMCU version
Current master (1.5.1)

### Hardware
NodeMCU DevKit 1.0

pjsg, 2016-06-12:
I tried this (with the insecure variant) and after less than 10 invocations of the script, the amount of free heap stabilized (actually it went up and down by around 200 bytes on each invocation). I have no idea what is going on....

djphoenix, 2016-06-13:
@pjsg actually mqtt memory usage is very unstable in case of use secure connection. In unsecure mode possible workaround is reuse connection (both closed and failed can be reopened again). But secure variant causes exception reset in this case. So recreation causes leaks, and after lot of tries device resets with out-of-memory. This is prehistory of this testcase and report :)

djphoenix, 2016-06-19:
Fixed... something 😧 
Not really understood which of patches are really do work. Now diff is looks like that:
```patch
diff --git a/d4437a4 b/e4cbd75
index d4437a4..e4cbd75 100644
--- a/d4437a4
+++ b/e4cbd75
@@ -125,6 +125,7 @@ static void mqtt_socket_disconnected(void *arg)    // tcp only
       mud->pesp_conn = NULL;
     }
 
+    mud->connected = false;
     luaL_unref(L, LUA_REGISTRYINDEX, mud->self_ref);
     mud->self_ref = LUA_NOREF; // unref this, and the mqtt.socket userdata will delete it self
   }
@@ -156,6 +157,11 @@ static void mqtt_socket_reconnected(void *arg, sint8_t err)
     pesp_conn->proto.tcp->local_port = espconn_port();
     socket_connect(pesp_conn);
   } else {
+    if (mud->secure) {
+      espconn_secure_disconnect(pesp_conn);
+    } else {
+      espconn_disconnect(pesp_conn);
+    }
     mqtt_socket_disconnected(arg);
   }
   NODE_DBG("leave mqtt_socket_reconnected.\n");
@@ -804,6 +810,9 @@ static int mqtt_delete( lua_State* L )
     c_free(mud->pesp_conn);
     mud->pesp_conn = NULL;    // for socket, it will free this when disconnected
   }
+  while(mud->mqtt_state.pending_msg_q) {
+    msg_destroy(msg_dequeue(&(mud->mqtt_state.pending_msg_q)));
+  }
 
   // ---- alloc-ed in mqtt_socket_lwt()
   if(mud->connect_info.will_topic){
@@ -873,8 +882,7 @@ static sint8 socket_connect(struct espconn *pesp_conn)
 #ifdef CLIENT_SSL_ENABLE
   if(mud->secure)
   {
-      espconn_secure_set_size(ESPCONN_CLIENT, 5120); /* set SSL buffer size */
-      espconn_status = espconn_secure_connect(pesp_conn);
+    espconn_status = espconn_secure_connect(pesp_conn);
   }
   else
 #endif
@@ -884,7 +892,7 @@ static sint8 socket_connect(struct espconn *pesp_conn)
 
   os_timer_arm(&mud->mqttTimer, 1000, 1);
 
-  NODE_DBG("leave socket_connect.\n");
+  NODE_DBG("leave socket_connect, heap = %u.\n", system_get_free_heap_size());
 
   return espconn_status;
 }
@@ -971,22 +979,21 @@ static int mqtt_socket_connect( lua_State* L )
     return luaL_error(L, "already connected");
   }
 
-  if(mud->pesp_conn){   //TODO: should I free tcp struct directly or ask user to call close()???
-    mud->pesp_conn->reverse = NULL;
-    if(mud->pesp_conn->proto.tcp)
-      c_free(mud->pesp_conn->proto.tcp);
-    mud->pesp_conn->proto.tcp = NULL;
-    c_free(mud->pesp_conn);
-    mud->pesp_conn = NULL;
+  struct espconn *pesp_conn = mud->pesp_conn;
+  if(!pesp_conn) {
+    pesp_conn = mud->pesp_conn = (struct espconn *)c_zalloc(sizeof(struct espconn));
[... 70 lines cut ...]
+      espconn_status = espconn_send(mud->pesp_conn, temp_msg->data, temp_msg->length);
+      if(mud->pesp_conn->proto.tcp->remote_port || mud->pesp_conn->proto.tcp->local_port)
+        espconn_status |= espconn_disconnect(mud->pesp_conn);
+    }
   }
+  mud->connected = 0;
+  
+  while (mud->mqtt_state.pending_msg_q) {
+    msg_destroy(msg_dequeue(&(mud->mqtt_state.pending_msg_q)));
+  }
+
   NODE_DBG("leave mqtt_socket_close.\n");
 
   if (espconn_status == ESPCONN_OK) {
@@ -1619,6 +1619,7 @@ static const LUA_REG_TYPE mqtt_map[] = {
 int luaopen_mqtt( lua_State *L )
 {
   luaL_rometatable(L, "mqtt.socket", (void *)mqtt_socket_map);  // create metatable for mqtt.socket
+  espconn_secure_set_size(ESPCONN_CLIENT, 4096);
   return 0;
 }
```

Now it works, but maybe need to be reviewed and filtered out unnecessary changes.
@pjsg @funshine I assume you are know better how this module works...

djphoenix, 2016-06-19:
Also fixed reconnection issue with secure connection. Now `:connect()` after `:close()` works (for same settings, not tried with different).

marcelstoer, 2016-07-24:
Yury, can you provide a PR for this?

djphoenix, 2016-07-24:
Yep, but need some refactoring and testing. I was want reaction from @pjsg & @funshine, but no way...

djphoenix, 2016-08-08:
Right, according to #1406 I consider this patch stable, so will make PR today.

PR #1445 (2016-08-14): MQTT memory / connection issues fix
Fixes #1349, #1406.
closed 2016-08-14 by marcelstoer