nodemcu/nodemcu-firmware #1351: Drop the examples folder?

marcelstoer, 2016-06-11:
I only recently looked at the content we have in the `examples` folder. Is it (still) worth keeping?

I find [`fragment.lua`](https://github.com/nodemcu/nodemcu-firmware/blob/master/examples/fragment.lua) in particular a bit odd. It appears to be a collection of unrelated, undocumented and unmaintained pieces of code for the initial set of NodeMCU modules. Some may not even work with the current firmware anymore. In its current form that file probably is of very little value to anyone.

devsaurus, 2016-06-11:
I agree, drop the folder but salvage `init.lua`. Guess it's the only reference we have atm, right?

marcelstoer, 2016-06-11:
> Guess it's the only reference we have atm, right?

You mean we don't explain the purpose of it enough (i.e. at all) in the docs? It's mentioned a few times in @TerryE 's "[Lua Developer FAQ](https://nodemcu.readthedocs.io/en/dev/en/lua-developer-faq/)". That page btw references an example from the aforementioned `fragment.lua`...

> drop the folder but salvage `init.lua`

I don't really see the point of that `init.lua` with a countdown. _If_ we want to keep an `init.lua` somewhere somehow I'd prefer if it showed something like this:

```lua
--load credentials, SSID and PASSWORD declared and initialize in there
dofile("credentials.lua")

function startup()
    if file.open("init.lua") == nil then
      print("init.lua deleted")
    else
      print("Running")
      file.close("init.lua")
--  dofile(myfile.lua)
    end
end

--init.lua
print("set up wifi mode")
wifi.setmode(wifi.STATION)
wifi.sta.config(SSID,PASSWORD)
wifi.sta.connect()
tmr.alarm(1, 1000, 1, function() 
    if wifi.sta.getip()== nil then 
        print("IP unavaiable, Waiting...") 
    else 
        tmr.stop(1)
        print("Config done, IP is "..wifi.sta.getip())
        print("You have 5 seconds to abort Startup")
        print("Waiting...")
        tmr.alarm(0,5000,0,startup)
    end 
 end)
```
Source: https://sourceforge.net/projects/nodemcu/. Side note: I find that project a bit problematic because it uses "our" brand name, files seem somehow replicated at https://github.com/ckuehnel/NodeMCU-applications..

TerryE, 2016-06-11:
`init.lua` is a standard Lua feature.

devsaurus, 2016-06-11:
I'm fine with any `init.lua` that prevents bricking the firmware as I've supported too many people with that problem in the past. Just keep an example somewhere.

TerryE, 2016-06-12:
I must admit i just stick a 3 sec tmr delay in mine when I am testing.  This is easily enough to paste in a file.remove() command to delete it if I am stuck in a reboot loop.

marcelstoer, 2016-06-12:
All right, here's my proposal:
- drop the folder
- add a small chapter about `init.lua` to https://github.com/nodemcu/nodemcu-firmware/blob/master/docs/en/upload.md showing something like I posted previously
- adjust the "Lua Developer FAQ" to replace references to `fragment.lua` with references to respective content in `lua_examples`

TerryE, 2016-06-12:
@marcelstoer, now you are making me feel guilty!! :)  The FAQ as published is woefully out of date and I really need to finish my rewrite, but there's not enough hours in the day.  Building your own house is amazingly time consuming even if you have an excellent builder doing a lot of the work!

PR #1353 (2016-06-13): Drop examples folder
Fixes #1351.
closed 2016-06-13 by marcelstoer