nodemcu/nodemcu-firmware #1356: MQTT publish message character limit

GelirDurum, 2016-06-13:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
 I worked on MQTT client and publish messages. Message was bigger then 1015 character and topic name was "Test". Messages didnt publish and i didnt saw any error messages.
### Actual behavior

### Test code
```
 topic = "Test"
clientid = node.chipid()
port = 1883
url = "broker.hivemq.com"
mclient = mqtt.Client(clientid,0,id,pass)

function sendmqtt (message)
    mclient:publish(topic,message,0,0, function(client) print("sent: "..message) end)
    collectgarbage()  
    print(string.len(message))
end

function subscribe ()
if mqttconnectflag then
    mclient:subscribe(topic,0, function(client) print("subscribe success") end)  
    mclient:on("message", function(clientid, topic, data) 
      print(topic .. ":" ) 
      if data ~= nil then
       print(string.len(data))
      end
    end)
end
end
function mqttconnect()
    mclient:connect(url, port, 0,1, function(client) print("connected")
    mqttconnectflag = true 
    subscribe()
    end, 
    function(client, reason) 
mqttconnect()
print("failed reason: "..reason) end)
end

mqttconnect()
function test(value)
ctr=0
a='a'
tmr.alarm(0, 10, 1, function() 
a=a..'a'
ctr=ctr+1
print(ctr)
collectgarbage()  
if ctr == value then
tmr.stop(0)
sendmqtt(a)
end
end)
end

```

 I saw char limit of publish messages was 115 by use this code. When i change  broker adress i saw same result. When i changed  topic name to "ArTest_MQTTLimit" limit change to 105.

### NodeMCU version
dev branch built on 11.04.2016
### Hardware
ESP8266_12F and usual connections

8<------------------------ END BUG REPORT -------------------------------------

marcelstoer, 2016-07-24:
> I saw char limit of publish messages was 115 by use this code. When i change broker adress i saw same result. When i changed topic name to "ArTest_MQTTLimit" limit change to 105.

Which suggests topic + message can have at most 120 chars?

stromnet, 2018-12-01:
This is most likely fixed with #2571, please give `dev` branch a try!

closed 2018-12-01 by marcelstoer