nodemcu/nodemcu-firmware #1360: AUTH_OPEN not definde?

plushvoxel, 2016-06-14:
### Expected behavior
```lua
> cfg={}
> cfg.ssid="myssid"
> cfg.pwd="mypassword"
> cfg.auth=AUTH_OPEN
> for k,v in pairs(cfg) do print(k.." : "..v) end
pwd : mypassword
ssid : myssid
auth : 0
```
### Actual behavior
```lua
> for k,v in pairs(cfg) do print(k.." : "..v) end
pwd : mypassword
ssid : myssid
```
### Test code
```lua
> =AUTH_OPEN
nil
> =wifi.AUTH_OPEN
nil
> =wifi.ap.AUTH_OPEN
nil
```
### NodeMCU version
NodeMCU 1.5.1 build on own machine without errors

### Hardware
ESP8266-12E

oyooyo, 2016-06-14:
That indeed appears to be a bug in the documentation, the correct identifiers seem to be
```
wifi.OPEN
wifi.WPA_PSK
wifi.WPA2_PSK
wifi.WPA_WPA2_PSK
```

marcelstoer, 2016-06-15:
It's a bit misleading because `AUTH_OPEN` is the C-side of Lua's `OPEN` as per https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/wifi.c#L1295. Those values are used both at http://nodemcu.readthedocs.io/en/dev/en/modules/wifi/#wifiapconfig and on http://nodemcu.readthedocs.io/en/dev/en/modules/enduser-setup/#example.

commit ed679a72 (2016-06-21): Correct WiFi auth constants in docs, fixes #1360
closed 2016-06-21 by marcelstoer