nodemcu/nodemcu-firmware #1363: Feature Request: Local socket Information

sebcologne, 2016-06-15:
### Missing feature
There is no way to get the receiving port-number of a tcp socket. Basically a getpeer() function, that returns the sockets local port.
See an example in python:[https://docs.python.org/2/library/socket.html#socket.socket.getsockname]( https://docs.python.org/2/library/socket.html#socket.socket.getsockname)

### Justification
NAT traversal in a DS-Lite network with unsymmetric NATs (Hole Punching).
You need the receiving port number to set up a new server that listens on that port for incoming connections, which are going through the "hole".

Why you ask?
Well, a popular German internet provider does not include a publicly accessible IPv4-adress to your sweet high-speed fiber connection. Instead you share your IPv4 adress with a bunch of other people.
A cheap way to eliminate this Problem is to use (preferrably TCP) Hole-Punching. ( Or use IPv6 with Devices that support it).

Also because I feel like a mad hacker every time I punch holes through Firewalls...

### Workarounds
Polling Data frequently from your server....

djphoenix, 2017-01-19:
As in current `dev` docs, after my work on `net` module there is [`getaddr`](http://nodemcu.readthedocs.io/en/dev/en/modules/net/#netsocketgetaddr) method for `socket`.

I think we may close it (so add `2.0.0` milestone because currently changes are in `dev` only)

/cc @marcelstoer

marcelstoer, 2017-01-19:
Thanks Yury!

closed 2017-01-19 by marcelstoer