nodemcu/nodemcu-firmware #1364: Please add the irsend lib to the nodemcu firmware！thanks！！！

codywon, 2016-06-15:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

### Actual behavior

### Test code
Provide a minimal and isolated test which will reproduce the problem.
```Lua
-- add code here
```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

marcelstoer, 2016-06-16:
You stand a much better chance seeing your feature request implemented if you fill in the form.

marcelstoer, 2016-06-29:
Not the slightest effort by the originator to even provide a useful description -> closing.

closed 2016-06-29 by marcelstoer