nodemcu/nodemcu-firmware #1365: .bin file for 1.5.1

SilentAce, 2016-06-16:
Hi,

I am pretty new to nodeMCU, ESP8266 and Lua. You already have bin file attached for version 0.9.6. Could please upload a .bin file for for latest version. Or if i can make a .bin file from the source code myself could you point me to the right tutorials. Thanks.

oyooyo, 2016-06-16:
The various possibilities of creating a binary firmware image are described here: https://nodemcu.readthedocs.io/en/dev/en/build/

To sum it up: To get started, you should simply use this web-based service: http://nodemcu-build.com/

It's by far the easiest and way to get a custom, up-to-date binary firmware image. Later on, you may want to look at building a firmware image from source.

SilentAce, 2016-06-16:
Thank you for the links and your help. Hope to build something awesome soon :).

oyooyo, 2016-06-16:
I'm sure you will, ESP8266 and nodemcu **are** awesome :)
Don't forget to close the issue.

closed 2016-06-16 by SilentAce

prithwirajbose, 2017-07-08:
I have tried building a firmware from nodemcu-build.com, the firmware loaded well in ESP however, the ESPlorer IDE never recognizes the firmware. Is there any way to update the definition of ESPlorer's firware recognition capability? As a result, ESPlorer behaves most weird than ever with meaningless errors while uploaded code (init.lua) to ESP. It changes my code and throws error. Isn't there a way to get a stable, pre-built firmware with basic modules? Very strange to see no direct link to download binaries anywhere on the cover page, everyone is linking this cloud platform link for making own build...

oyooyo, 2017-07-09:
@prithwirajbose ESPlorer's automatic firmware recognition has never worked for me, and I've witnessed quite a few people being confused about the firmware recognition feature not working. On the other hand, I've never come across any actual problem resulting from this. I don't know what exactly this firmware recognition feature was ever supposed to be useful for - ESPlorer works fine without it, and the feature can simply be disabled in the program's settings. So I'd personally suggest you simply disable the feature.
If ESPlorer is showing weird behaviour, I'd suppose that they do not really stem from the failing firmware recognition feature, there's probably another problem.

As for the missing pre-build firmware binary: In earlier versions, NodeMCU always provided such a pre-build firmware binary for download. But apparently, at some point they decided to no longer provide such files. The cloud platform for creating customized binaries is a great feature, but I too consider it a bad decision to no longer additionally provide at least one pre-built firmware binary.
My own fork of the nodemcu-firmware (https://github.com/oyooyo/nodemcu-firmware) therefor continues to provide such a file for download. It's basically just a pre-built firmware binary of the regular nodemcu firmware with most modules enabled and support for longer (63 character) filenames. It's not the outdated 1.5.4.1 firmware though.

OlairII, 2017-11-17:
@prithwirajbose I was having the same problem. You must use a baudrate of  115200 when you are working with the new versions of NodeMCU firmware. Just set baudrate to 115200 on ESPlorer and it should work fine.