nodemcu/nodemcu-firmware #1368: ESP12 dev/integer : Heap keeps decreasing after dsleep option 4

cendev, 2016-06-20:
I'm using nodemcu dev branch with the commit of f2d01957322c81387e902a8937cd749a696dfbf8 integer firmware and my problem is when my ESP-12 module wakes after node.dsleep with option 4, even without an init.lua my heap starts to decrease till hits the rock bottom and reboots itself.

custom build by frightanic.com
	branch: dev
	commit: f2d01957322c81387e902a8937cd749a696dfbf8
	SSL: false
	modules: crypto,dht,file,gpio,http,net,node,tmr,uart,wifi
 build 	built on: 2016-06-20 05:15
 powered by Lua 5.1.4 on SDK 1.5.1(e67da894)

Is this a normal behaviour ?

marcelstoer, 2016-06-21:
What's the Lua code you run/ran?

marcelstoer, 2016-06-21:
I believe you that `init.lua` is empty but where does "node.dsleep with option 4" come from? How was that issued? Is it the only statement or did you before init WiFi? If so, how?

karrots, 2016-06-21:
I also have an open issue on this topic. See #1225. @cendev is your unit stuck in the same state as I observed?

marcelstoer, 2016-06-22:
Thanks @karrots, that really sounds like a dup.

closed 2016-06-26 by marcelstoer