nodemcu/nodemcu-firmware #1369: node.restart() after set the Wifi Config causes crash

Azrrael-exe, 2016-06-20:
Hello Everyone

I am trying to create a custom interface to set up my node using an standar web server in AP mode, and after get the credentials switch to Station mode and reset the module, here is my code:

### Test code
.
```Lua
        wifi.setmode(wifi.STATION)
        ssid = arg["ssid"] -- SSID
        psk = arg["psk"] -- Password
        wifi.sta.config(ssid,psk)
        wifi.sta.connect()
        print("Node restart!");
        node.restart();
```

### Expected behavior

In the previous build of the firmware 0.9.6 it used to work perfectly

### Actual behavior

Now, with an autho generated build with the Node module, the the result is this:

```Lua
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 29228, room 16 
tail 12
chksum 0xc3
ho 0 tail 12 room 4
load 0x3ffe8000, len 2232, room 12 
tail 12
chksum 0xfc
ho 0 tail 12 room 4
load 0x3ffe88b8, len 8, room 12 
tail 8
chksum 0x60
csum 0x60
don't use rtc mem data
```

### NodeMCU version

```Lua
NodeMCU custom build by frightanic.com
\0x09branch: master
\0x09commit: cdaf6344457ae427d8c06ac28a645047f9e0f588
\0x09SSL: false
\0x09modules: cjson,file,gpio,http,i2c,mqtt,net,node,tmr,uart,wifi
 build \0x09built on: 2016-06-20 22:26
 powered by Lua 5.1.4 on SDK 1.5.1(e67da894)

```

Anyone have a solution or a clue to solve this?

Thanks in advance

oyooyo, 2016-06-21:
You'll probably have to wait for someone more experienced to come around, just a few thoughts. My understanding is that your test code basically does the following:

1. Configure station mode via `wifi.sta.config(ssid,psk)`. Since the "auto" argument in that function call is omitted, this will automatically initiate the connect process as well (see [documentation of wifi.sta.config](https://nodemcu.readthedocs.io/en/dev/en/modules/wifi/#wifistaconfig))
2. With the connect process just being initiated, but not connected yet, initiate the connect process once more via `wifi.sta.connect()` (which seems unnecessary at best, and might eventually even cause unexpected behaviour)
3. With the connect process still just being initiated, but far from actually being connected yet, immediately restart the chip via `node.restart()`, which causes the usual boot message posted under "actual behaviour".

I'm failing to see the intention behind that test code, what you're actually trying to achieve here (with the node.restart() call etc.) and thus how/why the actual output differs from what you expected. To me the output looks like nodemcu-firmware is simply doing what you told it to do.

marcelstoer, 2016-06-21:
@oyooyo great reply, spot-on!

closed 2016-06-21 by marcelstoer

Azrrael-exe, 2016-06-21:
Thanks @oyooyo for the Quick responses, for now my problem is solved, you have reason, the restart of the node is not necessary.

My idea behind that restart was just restart the whole program, it works normally connected as Station mode, and the AP mode an exception, my idea was just set the credentials and start again.

The new code looks like this:

```Lua
        wifi.setmode(wifi.STATION)
        ssid = arg["ssid"]
        psk = arg["psk"]
        wifi.sta.config(ssid,psk)
        tmr.alarm(1, 5000, 1, initialize) -- function to initialize the program
```
however the node.restart() instruction is causing the output mentioned before, even without the wifi instructions.

marcelstoer, 2016-06-21:
This is not the right place to discuss such issues -> http://nodemcu.readthedocs.io/en/latest/en/support/