nodemcu/nodemcu-firmware #1370: Feature Request: Date/time conversion for rtctime.get()

devsaurus, 2016-06-22:
### Missing feature
Conversion of a timestamp "seconds since Unix epoch" into calendar items year/month/day/hour/min/sec.

### Justification
Seconds since epoch can't be handled easily in a human readable way.

### Workarounds
Hour & second components can be calculated by simple formulas in Lua land. Date is more complex and less efficient but error prone.

### Proposal
A respective conversion routine (gmtime_r) is available from [dietlibc](http://www.fefe.de/dietlibc/) and I would add it with a Lua binding to the code base.
`tm = xxx.unix2date(stamp, timezone)`
Returns a table with entries
- `year`
- `month`
- `day`
- `hour`
- `minute`
- `second`

Should this function go into rtctime, node or a dedicated module?

marcelstoer, 2016-06-23:
Only rtctime has a meaningful representation for "seconds since Unix epoch", right (unless queried from external service)? If so, then it should be added to that module IMO.

pastukhov, 2016-06-24:
This issue has been discused here #603

PR #1386 (2016-07-19): Add rtctime.epoch2cal()
Fixes #1370.

- [x] This PR is compliant with the [contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

Conversion of Unix timestamp to calendar date & time as described in the related issue.

Committers supporting this PR: Marcel
closed 2016-07-24 by marcelstoer