nodemcu/nodemcu-firmware #1371: wifi reset

bastos1505, 2016-06-26:
Hi guys, I'm using ESP12 and I'm facing a problem with wifi connection. I connected my ESP in my router and everything was ok up to my Set-Top-Box (that as well is connected in same router) be turned ON.
All the pins are OK and the VCC is very fine. I can't have a LOG because the ESP just resets.

The reset msg is: rst #1 (normal boot). There is no glitches in VCC. I measured with a scope and didn't see anything.

Facts:

When I disconnect the ESP from my router, the problem doesn't happens;
The problem just happens when my Set-Top-Box is booting;
Even with a big data traffic, there's no problem;
I can't have a LOG, because the ESP just resets;
I'm using nodemcu 1.5.1 integer (Master) WIFI, UART, FILE, GPIO, HTTP, NODE, NET, TIMER.

bastos1505, 2016-06-26:
There are many issues that don't follow your template. Sorry. 
About the evidences, I can film it for you. That's the point! The ESP just resets. There is no messages!

FlyerBoltz, 2016-06-27:
Hello there,  I think I have solved the issue.  It solved for me by using the original node mcu flasher and flashing blank. Bin to every address available first.  Once done.  Upload the firmware files to their respective addresses and boom. The error disappeared the error which I had got was a wdt reset error with some "stack".

bastos1505, 2016-06-28:
Hi guys, I found a weird thing in my LAN. When my set-top-box is requesting DHCP/ARP, the Set-Top-Box doesn't issues "GRATUITUOS ARP REQUEST". After this, the ESP resets.
If you want, I can send the wireshark file to help you.
The STB performs the folowing steps:
1)DHCP Discovery;
2)DHCP Offer;
3)DHCP Request;
4)DHCP ACK;
5)Issue ARP Broadcast asking for his IP; (**after this point all the other devices issue GRATUITOUS**)
6)Issue ARP Broadcast informing its IP is at MAC;
7)Issue ARP Broadcast requsest asking for Router IP;
8) Crash.
Thanks,

bastos1505, 2016-07-16:
Hello guys, I have found the problem. 
The problem was happening because the dhcp_msg struct uses only 312 bytes for options (uint8_t options[312]). But, there are many devices that send more than 312 bytes. So, the right way to do that is use option as uint8_t * and allocate dynamically in the handle_dhcp function.

devsaurus, 2016-07-16:
> The problem was happening because the dhcp_msg struct uses only 312 bytes for options

I haven't looked into the lwip code at all up to now, so my conclusions might be wrong. But a quick review of `handle_dhcp()` in `app/lwip/app/dhcpserver.c` confirms the described problem theoretically:
The code allocates a fixed size for dhcps messages assuming the options field to have 312 bytes length. It then copies payload from up to 2 messages into this heap region without further range checks. If the payload chunks are longer than assumed then this will inject / overwrite data in the following memory. Nasty :disappointed:.
Maybe there's an implicit limit on protocol level, but code should never break out of it's own region IMO.

> So, the right way to do that is use option as uint8_t * and allocate dynamically in the handle_dhcp function.

Can please you provide the code (preferably as a fork)?

On a general level I assume that the lwip code is inherited from [Espressif's release](http://bbs.espressif.com/viewtopic.php?f=46&t=1913), right? Any issue & fix would need to communicated to them to avoid deviations from upstream?

bastos1505, 2016-07-18:
Yes I can. But, I need help to use git or another kind of CVS. I'm a Hardware man :) So, I don't kown too much about SW tools.
If you want, I can send you the files. No problem. Otherwise, I'll need help to upload the files.
BR,

marcelstoer, 2016-07-18:
@bastos1505 see if https://github.com/nodemcu/nodemcu-firmware/blob/master/CONTRIBUTING.md helps you get started.

devsaurus, 2016-07-18:
I'm experimenting with a possible fix for this issue at https://github.com/devsaurus/nodemcu-firmware/tree/dhcp_fix
The analysis based on @bastos1505's description is backed by my tests where crafted DHCP packets are sent to the ESP in SOFTAP mode: Standard size packets are ok while oversized packets cause an immediate crash.

@bastos1505 please have a look at my dhcp_fix branch and check whether it fixes your real-world problem. Also let me know if you require a pre-compiled image for testing.

marcelstoer, 2016-07-24:
How does #1412 come into play here?

devsaurus, 2016-07-24:
> How does #1412 come into play here?

My approach in the [referenced branch](https://github.com/devsaurus/nodemcu-firmware/tree/dhcp_fix) is to stick with the default size and avoid any buffer overflow. This will break the protocol if a client sends more options than fit into 312 bytes.

#1412 on the other hand extends the size according to the received packet which will most likely cover more real-world corner cases. The drawback is that clients sending excessively large options will crash the firmware again.

I've flagged this topic already to Espressif with reference to this issue and would like to wait for a solution from them. If they don't react we should go for a local fix based on #1412 with additional sanity checks.

PR #1421 (2016-08-02): Fix DHCP server heap corruption
Fixes #1371.

- [x] This PR is compliant with the [contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [ ] The code changes are reflected in the documentation at `docs/en/*`.

This PR fixes the DHCP server heap corruption in case an oversized package is sent to the server.
The fix is done as per recommendation from Espressif:
> To avoid crash in DHCP big packages modify option length from 312 to MTU - IPHEAD(20) - UDPHEAD(8) - DHCPHEAD(236) [... cut]
closed 2016-08-02 by marcelstoer