nodemcu/nodemcu-firmware #1372: Feature request: HomeKit adaptor module

djphoenix, 2016-06-27:
HomeKit is Apple technology for manage IoT devices. It seems be very good to have a module to connect ESP devices to this platform because it provides remote control with unified interface (e.g. MyHome app) and voice control with Siri.

Realization might be done with porting solutions like [HAP-NodeJS](https://github.com/KhaosT/HAP-NodeJS) or [Personal-HomeKit-HAP](https://github.com/etwmc/Personal-HomeKit-HAP).

marcoskirsch, 2016-06-27:
I don't know the technical feasibility of this but it would sure be cool.

someburner, 2016-06-27:
I haven't messed around with SSL on the ESP much, but I've gotten the ESP to show up on Avahi which is mentioned in the Personal-HomeKit-HAP project. Of course that's just the device discovery.. The device pairing looks fun...

marcoskirsch, 2017-08-20:
FWIW: 
https://github.com/HomeACcessoryKid/ESP8266-HomeKit-Demo
https://github.com/HomeACcessoryKid/ESP8266-HomeKit

So it _can_ be done on ESP8266. Whether it can be done along with NodeMCU remains to be seen.

DISCLAIMER: I didn’t try those projects. If anyone did I’d love to hear your experience.

rodvlopes, 2019-01-01:
Hi, I tried the ESP8266-HomeKit-Demo and it works just perfect, I could control the ESP8266 with Siri voice commands. The ESP8266-HomeKit-Demo is built on esp-open-rtos. It would be great to see the Homekit working with the simplicity of the NodeMCU.

nwf, 2019-01-02:
Please see #1010; is someone stepping up to try to merge the homekit (demo) work into nodemcu proper?

closed 2020-01-11 by stale[bot]

ceifa, 2024-01-05:
Anyone had success on using it with nodemcu? Would be great to have a nodemcu package of it published.