nodemcu/nodemcu-firmware #1374: Expose random number at 0x3FF20E44

marcelstoer, 2016-06-27:
### Missing feature
Random number generator. According to http://esp8266-re.foogod.com/wiki/Random_Number_Generator the easiest seems to be to offer a read method for 0x3FF20E44. Don't the crypto functions have something like that internally?
### Justification
Some guy over at esp8266.com built his own firmware with the disabled Lua math module enabled and then complained about bad results. Ok, not a good enough justitication but stil...
### Workarounds
Hand-woven randomness or use https://www.random.org/clients/http/.

devsaurus, 2016-06-27:
Seems to be related to #1205.

marcelstoer, 2016-06-27:
Thanks Arnim, it _did_ sound vaguely familiar but I only searched all issues for 'random' (i.e. not the PRs). So, how to go from here? Shall we revive Johny's PR and close this issue?

marcelstoer, 2016-06-27:
Related: why does one of our DHT examples contain `math.floor`, http://nodemcu.readthedocs.io/en/dev/en/modules/dht/#example? This contradicts http://nodemcu.readthedocs.io/en/dev/en/lua-developer-faq/#how-is-nodemcu-lua-different-to-standard-lua

devsaurus, 2016-06-27:
> Shall we revive Johny's PR and close this issue?

The discussion seemed to end somewhat inconclusive with 3 different approaches:
- `crypto.rnd32()` as proposed initially
- `math.random()` alternative by Philip with specific implications
- direct binding of `os_[get_]random()`

Not sure atm what should be the final solution.

djphoenix, 2016-06-27:
*// Sorry about my IMHO, but...*
I think **the final solution** should be chosen by maintainers. Actually right now at both master and dev we have not *any* of RNG/PRNG. #1205 was good solution, but closed...

pjsg, 2016-06-28:
I feel that (at least) the builtin lua random number functions should use the platform random number generator (on the assumption that expressif/xtensa did a good job). It might be that this is the only implementation. 

a 32-bit random number can be got with math.random(-0x80000000, 0x7fffffff)

Actually, I realize that my implementation in the other PR doesn't handle (in the floating point firmware) ranges like    

math.random(0, 0x7fffffffffff)

jmattsson, 2016-08-11:
To summarise my concerns in that area:
- Making `math.random()` use the HWRNG removes the PRNG feature; I think both should be available, through whatever approach we use (even if it's an ugly `math.setrng(math.HWRNG || math.PRNG)`)
- TriAnMan raised a good point about the ability to get a whole block of random bytes, I'd like to see that included somehow.
- Lua got the range bounds wrong for `math.random()`, picking `1 <= r <= n` aka `[1,n]` rather than `1 <= r < n` aka `[1,n)`.

pjsg, 2016-08-11:
INterestingly enough, the standard lua library gets math.random wrong s well when the range is very large. In the two argument form, it treats both values as signed 32-bit integers. 

```
> print (0xb0000000 > 0x70000000)
true
> print(string.format("%x", math.random(0x70000000, 0xb0000000)))
stdin:1: bad argument #2 to 'random' (interval is empty)
stack traceback:
	[C]: in function 'random'
	stdin:1: in main chunk
	[C]: ?

```

I have a working implementation of math.random that uses the hardware random number generator, but the approach is not pretty (it doesn't minimize the amount of change to the stock lua interpreter). I can try and clean it up.

truhlikfredy, 2016-10-08:
Any implementation would be nice

computerlyrik, 2016-12-10:
@pjsg  +1 for a pull request

pjsg, 2016-12-10:
There are two options -- one to change the existing `math.random` to return truly random numbers. The other is to add `node.random` which has an identical API to `math.random` but deals with the true random numbers from the hardware.

Opinions?

truhlikfredy, 2016-12-10:
I would go with node.random, just in case somebody relies on pseudo-random numbers, their behaviour wouldn't change.

djphoenix, 2016-12-11:
I think good point is to provide `node.random_buf` (with length-argument, name is discussable) that just returns string with random bytes.

PR #1665 (2016-12-26): Added node.random implementation
Fixes #1374.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

This adds node.random which behaves the same way that math.random does *except* that it uses the hardware random number generator for (more) unpredictably random numbers. It also gets large [... cut]
closed 2016-12-26 by marcelstoer