nodemcu/nodemcu-firmware #1375: before load init.lua all GPIO are OUTPUT HIGH

ozyart, 2016-06-27:
8<------------------------ BUG REPORT -----------------------------------------

Before init.lua load, all GPIO OUTPUT are HIGH.
I need all GPIO in LOW before load init.lua

I hope one solution, thanks

### NodeMCU version
NodeMCU 0.9.5 build 20150318  powered by Lua 5.1.4

### Hardware
ESP8266 12E

8<------------------------ END BUG REPORT -------------------------------------

devsaurus, 2016-06-27:
My guess would be that all GPIOs power-up with their pull-up resistors enabled for safe operation. Even though GPIOs could be set to drive LOW quite early in the start-up process, there would be always this initial phase.

How long can your use case tolerate that all GPIOs are HIGH before they're driven LOW by software?

ozyart, 2016-06-27:
I test it with diferent resistors and diodes.

When I turn the esp12 UP, All GPIO are HIGH during 1 second.
When read init.lua everyone go to LOW.
I need  this always start in LOW.

ozyart, 2016-06-27:
All GPIO load with 1.73V, and if I don't define GPIO to OUTPUT this doesn't go to 0V

marcelstoer, 2016-06-28:
We had this a few times before e.g. #897. It's down to the hardware specs and the Espressif SDK. NodeMCU 0.9.x is no longer supported.

closed 2016-06-28 by marcelstoer

ozyart, 2016-06-28:
how I resolve that?

nickandrew, 2016-06-28:
Perhaps you could invert the logic in your hardware so starting high is OK.

ozyart, 2016-06-28:
I will invert the logic with software, I go try this with C code

ozyart, 2016-06-30:
last version work fine!

SupotSaeEa, 2016-09-20:
Use last version 1.5.4.1. The problem still occurred.