nodemcu/nodemcu-firmware #1382: Docs don't seem to state how to remove callbacks from net.socket()

jaggzh, 2016-07-07:
Prior to net.socket.close(), how can one remove a callback set with net.socket.on()?
I can examine the source to find out -- but I'm learning how to handle contributions / pull requests.

devsaurus, 2016-07-11:
> how can one remove a callback set with net.socket.on()?

@jaggzh as far as I can see, there's no method to remove callbacks which were registered with `net.socket.on()`. They're implicitly unregistered when the socket object is reaped during automatic garbage collection.

You should see a respective effect with
```lua
sck:close()
sck = nil   -- allows garbage collection to delete the object
```

marcelstoer, 2016-07-11:
IIRC some of our modules/methods allow you to pass `nil` for the callback to reset it. Assuming I understand `socket:on()` code correctly this won't work here, though.

devsaurus, 2016-07-14:
> IIRC some of our modules/methods allow you to pass nil for the callback to reset it.

Agree, the logic in `socket:on()` will only allow to set a callback but not to unregister an existing one.

Is this a use case? It could be added without breaking backwards compatibility IMO.

marcelstoer, 2016-08-07:
No feedback from the OP in a month -> closing.

closed 2016-08-07 by marcelstoer

jaggzh, 2016-08-12:
Sorry, I didn't know there was a question for me.  I didn't understand, "is this a use case?"  It's just that I registered some debug info for the connection, but didn't want it on an ongoing basis; however, when trying to find how to un-register it, I found it didn't seem possible.

jaggzh, 2016-08-12:
That is -- I didn't want to close the socket, and didn't want further connections to call my debug routine (and didn't want to register a separate dummy function to override it (if that works -- I've not checked if the events are added, or replaced).

devsaurus, 2016-08-12:
I wanted to figure out whether the missing unregister restricts you and other users in using the `net` module.
Is it worth the effort to add a way to unregister callbacks or is it just nice to have?