nodemcu/nodemcu-firmware #1384: TCP Server with SSL

jackasses, 2016-07-08:
The API from Espressif support a TCP-Server with SSL connection.

Is it possible or implemented to run this with nodemcu? 
I only see the options for the tcp-client.

djphoenix, 2016-07-09:
espconn layer is deprecated since announce of ESP-32 and RTOS-SDK. After migration (and integration of `mbedtls`) this functionality may be provided.

jackasses, 2016-07-11:
i found mbedtls that is working on the ESP-8266. The performance is not the best but i have to try it.

tdotu, 2016-08-26:
so, does this mean that the non RTOS-SDK version of nodemcu-firmware is "depricated"?

djphoenix, 2016-08-26:
Nope. Just espconn. Mbedtls can be used in NONOS SDK. But after first release of RTOS-based NodeMCU, NONOS versions will become outdated, so.

georeb, 2016-09-22:
Are there any advances on this? @marcelstoer 
Including secure connections for net.createServer() is a very high priority in my opinion.
It would open up a whole new world of usage for the ESP!

jmattsson, 2016-09-23:
There is a PR in the works to move the `net` module over to native lwIP, and after that I believe the plan is to switch over to mbedTLS to give us a newer/better SSL/TLS stack, which should hopefully allow us to do secure servers.

smcmurray, 2017-03-05:
v2.0 doesn't seem to allow tls server. Am I misunderstanding the docs?

HHHartmann, 2018-04-07:
It world be really great to have secured TCP comminucation from ESP8266 to ESP8266.

closed 2019-08-04 by stale[bot]