nodemcu/nodemcu-firmware #1387: Featured: Nodemcu need to setup a enviroment for automated tests

MarsTechHAN, 2016-07-11:
Now the project is getting bigger, as ESP8266 dont have a sandbox for each of the modules, we need to have a cross test when running Travis-CI. 

Basic work flow(during the CI Process):

>build the firmware->serch for not busy esp8266 module->update firmware for esp8266->run test script->return answer to CI -> end of build

djphoenix, 2016-07-11:
Hardware/Firmware test suites are very complex. Not sure we're needed to autotest modules because every time that changed something it will be community-tested on PR-based routine.
Also some of synthetic tests that we *may* write will not cover real-word situations. E.g. PWM will set-up and run, but will not generate actual signal.

MarsTechHAN, 2016-07-11:
0. Dont say those are complex/impossible... before acturally doing so : )
1. The target of this is mostly for basic function. like net./tmr./... to make sure everything works well.
2. the Community test is not a cover-oriented test, which means unless somebody find the issue(like memory leakage)  or they acturally touch the issue, none will  recongize them. And for example, someone run the script on master , but failed on dev, for nobody will know which part goes wrong(for so many PR has made). It is a good way to tracing everything（Also CI may results in a graph about one of the things we cared most-> mem usage）.
3. AT is the future, and we should have AT running on our project before it become too big to control the cross-module-bug or other potenial.

marcelstoer, 2016-07-11:
This is a dup of #516 which I created a year ago. I suggest we continue there.

closed 2016-07-11 by marcelstoer