nodemcu/nodemcu-firmware #1391: Feature request: RFID MFC522 integration into the firmware

roland-vachter, 2016-07-13:
Unfortunately, I found support for this popular, cheap RFID reader/writer for Arduino only.
It would be nice to port it into nodemcu as well, unfortunately I lack the knowledge to do that (programming on driver level).
Is this something that can be achieved?

NicolSpies, 2016-07-13:
Will be very helpful, RFID MFRC522 instead of MFC522.

roland-vachter, 2016-07-14:
Yes, correct, it was a typo. I'll change the title.

roland-vachter, 2016-07-14:
Any chance this to be included in a short term?

marcelstoer, 2016-07-14:
> Any chance this to be included in a short term?

Impossible to answer for an open-source project that people contribute their spare time to. Would be a nice addition though, indeed. 
I believe the Arduino library, which uses SPI, can serve as a really helpful blueprint.

cmisztur, 2016-08-19:
I am willing to help out with this.  @roland-vachter have you started anything?

roland-vachter, 2016-08-19:
@cmisztur that would be awesome! As I said, I lack the knowledge of C at this low level ... but I can help testing it if there will be any progress :)

piebat, 2016-08-24:
I'm interested too. At moment I'm trying to read just Card ID using the spi module.

samuel4711, 2016-09-09:
@piebat you try to read the Card ID ? did you get it work ? 
i found this one ...
http://www.instructables.com/id/WiFi-RFID-Reader
but for me it doesnt work :-( i try 2 esp8266 and nothing happens. i hope the get it work ...

piebat, 2016-09-10:
With arduino environment I just try once, with a ESP12E dev board (NodeMCU) and didn't work. I need to try better to see if depend on my setting. I try to port the function from library by Miguel Balboa in LUA, using spi module of NodeMCU firmware. I achieve to read the registers of PCD (it is the R522 chip) but I couldn't write them, so i couldn't achieve any function related to the RFID CARD, at moment. I'm trying to just read the Card ID, nothing more sofisticate than this. As module I used a WROOM-02. It should not change compared with any ESP12E

spaccabit, 2017-01-04:
Hi,
I am a newbie. I would try to convert [this](https://github.com/miguelbalboa/rfid/blob/master/MFRC522.cpp) to native nodeMCU module.

I have some problem to map Arduino SPI methods to esp8266 api.

Can you help me to translate this code:

```C++
/**
 * Writes a number of bytes to the specified register in the MFRC522 chip.
 * The interface is described in the datasheet section 8.1.2.
 */
void MFRC522::PCD_WriteRegister(	byte reg,		///< The register to write to. One of the PCD_Register enums.
									byte count,		///< The number of bytes to write to the register
									byte *values	///< The values to write. Byte array.
								) {
	SPI.beginTransaction(SPISettings(SPI_CLOCK_DIV4, MSBFIRST, SPI_MODE0));	// Set the settings to work with SPI bus
	digitalWrite(_chipSelectPin, LOW);		// Select slave
	SPI.transfer(reg & 0x7E);				// MSB == 0 is for writing. LSB is not used in address. Datasheet section 8.1.2.3.
	for (byte index = 0; index < count; index++) {
		SPI.transfer(values[index]);
	}
	digitalWrite(_chipSelectPin, HIGH);		// Release slave again
	SPI.endTransaction(); // Stop using the SPI bus
} // End PCD_WriteRegister()

/**
 * Reads a number of bytes from the specified register in the MFRC522 chip.
 * The interface is described in the datasheet section 8.1.2.
 */
void MFRC522::PCD_ReadRegister(	byte reg,		///< The register to read from. One of the PCD_Register enums.
								byte count,		///< The number of bytes to read
								byte *values,	///< Byte array to store the values in.
								byte rxAlign	///< Only bit positions rxAlign..7 in values[0] are updated.
								) {
	if (count == 0) {
		return;
	}
	//Serial.print(F("Reading ")); 	Serial.print(count); Serial.println(F(" bytes from register."));
	byte address = 0x80 | (reg & 0x7E);		// MSB == 1 is for reading. LSB is not used in address. Datasheet section 8.1.2.3.
	byte index = 0;							// Index in values array.
	SPI.beginTransaction(SPISettings(SPI_CLOCK_DIV4, MSBFIRST, SPI_MODE0));	// Set the settings to work with SPI bus
	digitalWrite(_chipSelectPin, LOW);		// Select slave
	count--;								// One read is performed outside of the loop
	SPI.transfer(address);					// Tell MFRC522 which address we want to read
	while (index < count) {
		if (index == 0 && rxAlign) {		// Only update bit positions rxAlign..7 in values[0]
			// Create bit mask for bit positions rxAlign..7
			byte mask = 0;
			for (byte i = rxAlign; i <= 7; i++) {
				mask |= (1 << i);
			}
			// Read value and tell that we want to read the same address again.
			byte value = SPI.transfer(address);
			// Apply mask to both current value of values[0] and the new data in value.
			values[0] = (values[index] & ~mask) | (value & mask);
		}
		else { // Normal case
			values[index] = SPI.transfer(address);	// Read value and tell that we want to read the same address again.
		}
		index++;
	}
	values[index] = SPI.transfer(0);			// Read the final byte. Send 0 to stop reading.
	digitalWrite(_chipSelectPin, HIGH);			// Release slave again
	SPI.endTransaction(); // Stop using the SPI bus
} // End PCD_ReadRegister()
```

digitalWrite -> platform_gpio_write

SPI.beginTransaction -> platform_spi_setup ??

SPI.transfer -> ??

devsaurus, 2017-01-04:
@spaccabit there's no direct mapping between Arduino's SPI methods and the platform functions here.

1. Don't call `platform_spi_setup()` in your module code but let the user configure the SPI bus from Lua scripts. Furthermore note that activating HSPI implicitly assigns Pin 8 to /CS, and it will toggle with every transaction. This should be avoided when setting up SPI:
```lua
spi.setup(1, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, 8, 8)
-- we probably won't be using the HSPI /CS line, so disable it again
gpio.mode(8, gpio.INPUT, gpio.PULLUP)
```
2. Don't forget to set `_chipSelectPin` to output at the beginning:
```c
platform_gpio_write(_chipSelectPin, PLATFORM_GPIO_HIGH);
platform_gpio_mode(_chipSelectPin, PLATFORM_GPIO_OUTPUT, PLATFORM_GPIO_FLOAT);
```
3. Use `platform_spi_send_recv()` instead of `SPI.transfer()`.
4. Skip `SPI.beginTransaction()` and `SPI.endTransaction()`.

That should do the trick.

In any case I'd recommend to verify the waveforms of CLK, MOSI, MISO, Pin8 and _chipSelectPin with a scope or logic analyzer to rule out a misconfiguration of the HSPI hardware.

NicolSpies, 2017-01-04:
Hi Giuseppe,

 

I have had the same requirement to use the RFID MFRC522 with the nodemcu and also found some Arduino code to be translated. I found that the following code on GitHub that has been translated already.

 

https://github.com/capella-ben/LUA_RC522/blob/master/RC522.lua

 

I have not had time to try it out. If you do please provide me with some feedback.

 

Regards

Nicol

marcelstoer, 2017-01-04:
Note that the intention of this issue is to provide a _native_ NodeMCU module rather than a Lua module.

NicolSpies, 2017-01-04:
Oops, sorry

spaccabit, 2017-01-09:
thank you all for the information

DiegoAscanio, 2017-01-15:
guys, a different aproach could be possible, since MFRC522 supports i2c and nodemcu have native support to this protocol.

I'll study MFRC522 datasheet and try to make MFRC522 work using i2c protocol

roland-vachter, 2017-07-20:
Hello,
I'm checking back here. Anyone had a chance to look into this one?

nickandrew, 2017-08-03:
I'd use a native module if somebody writes one. I've got an RC-522 gathering dust on a shelf. I should try the Lua module too.

curlyz, 2017-09-20:
I am working on porting this with I2C interface , coming soon , but is it i2c clock stretching a problem anymore with nodemcu ??
Still , Lua module , not native module.

divebomb10, 2017-11-21:
have you tried this>>>

http://www.esp8266.com/viewtopic.php?f=13&t=3435&sid=aa4c2b0196afdb2740a6555d5b91efdb&start=4

roland-vachter, 2017-11-22:
@divebomb10 the example you provided is Arduino, not for nodemcu.

divebomb10, 2017-11-22:
arduino what? do you mean ARDUINO IDE?

your question was, if the MFRC522 module can be port to a Nodemcu?

then that sketch is working for Nodemcu using ARDUINO IDE...

roland-vachter, 2017-11-22:
You can use either Arduino IDE with ESP8266, or Nodemcu. Both are a firmware (I think), but they cannot be used in the same time.
I think your mention of Nodemcu refers to the board, but that's ESP8266, nodemcu is the firmware that you install on it.
https://en.wikipedia.org/wiki/NodeMCU
But anyway, thanks for the link. If it could be ported into nodemcu as a native part of the firmware that would really help.

divebomb10, 2017-11-22:
nice one, so you want to use mfrc522 and the nodemcu board using LUA?

NicolSpies, 2017-11-22:
yes please

roland-vachter, 2017-11-22:
Yep, that's right. But the main idea of this issue is to have it embedded in the nodemcu firmware in native C, rather than a Lua lib (for performance).

divebomb10, 2017-11-22:
with that, i dont know how...

but i suggest that you use arduino ide for that, since all the information you can get online, is more on arduino language...

roland-vachter, 2017-11-22:
Well, this is an issue on nodemcu.
While some might prefer Arduino and C, others nodemcu and Lua. This is a feature request for nodemcu, which helps those who prefer Lua like me, it's closer to JavaScript than C.
So the feature request remains open.

marcelstoer, 2017-11-22:
@divebomb10 welcome to our community, did you realize whose party you just crashed? This is the NodeMCU _firmware_ repository i.e. a firmware with a Lua API. It's like you post to a Microsoft .Net forum to tell them to checkout how to solve problem X with Java.

If you had actually invested the time in reading the OP's description and some of the comments that followed it you would have noticed that it's been proposed before to port the Arduino code to this firmware. Your comment was off-topic and triggered lots of other OT comments. Anyone coming here will have to go through all of that again just to find out that half of the comments are irrelevant.

I'm tempted to delete this entire sub-thread (including my comment) but I'll sleep over it once again.

24540477zxc, 2018-10-10:
Two years have passed, this module is still no progress

closed 2019-10-19 by stale[bot]