nodemcu/nodemcu-firmware #1392: Bug: NodeMCU-Build custom rom vs Nokia 5110 LCD Driver

nguyenhuu43, 2016-07-13:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Hello, I want to connect between ESP-03 runs custom build from NodeMCU-Build and Nokia 5110 LCD PCD8544). I ran the code as following, but when it run to line
`pcd = u8g.pcd8544_84x48_hw_spi(cs, dc, res)` 
, the board send some un-readable text, and reset.

### Actual behavior

### Test code

```
function init_spi_display()
     local cs  = 8 -- GPIO15, pull-down 10k to GND
     local dc  = 4 -- GPIO2
     local res = 0 -- GPIO16

     spi.setup(1, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, 16, 0)

     pcd = u8g.pcd8544_84x48_hw_spi(cs, dc, res)
end
-- the draw() routine
function draw()
     pcd:drawStr( 5+0, 5+0, "Hello!")
end

init_spi_display()

draw()
```

### NodeMCU version

> NodeMCU custom build by frightanic.com
> 	branch: master
> 	commit: b580bfe79e6e73020c2bd7cd92a6afe01a8bc867
> 	SSL: false
> 	modules: file,gpio,http,i2c,mdns,mqtt,net,node,pwm,sntp,spi,tmr,u8g,uart,wifi
>  build 	built on: 2016-07-13 09:12
>  powered by Lua 5.1.4 on SDK 1.5.1(e67da894)

### Hardware

> ESP-03 with PL2303

8<------------------------ END BUG REPORT -------------------------------------

nguyenhuu43, 2016-07-13:
Screenshot as below
[http://www.upsieutoc.com/images/2016/07/13/Capture.jpg](url)

devsaurus, 2016-07-13:
What is the output of `print(node.bootreason())` after the crash?

BTW change the databits from 16 to 8 in `spi.setup()`.

nguyenhuu43, 2016-07-14:
Hi, I changed from 8 to 16 and 16 to 8, it makes non sense.
When I send node.bootreason, it return 2 (reset - software) and 6 (external reset), as following
http://www.upsieutoc.com/images/2016/07/14/Capture2.jpg

devsaurus, 2016-07-14:
I tried your code with the following results. Note that I don't have the PCD8544 hardware, there's nothing connected to HSPI.
- `pcd = u8g.pcd8544_84x48_hw_spi(cs, dc, res)` executes without problems
- Executing `pcd:drawStr( 5+0, 5+0, "Hello!")` in `draw()` crashes subsequently because no font is defined. Please add a `pcd:setFont()` before executing the first drawing routines.
- You're missing the firstPage() / nextPage() logic -> the text will only be drawn to a small portion.

> When I send node.bootreason, it return 2 (reset - software) and 6 (external reset)

This would suggest that one of the GPIOs used for SPI communication triggers an external reset, somehow.
- Any of these connected to nRST?
- Is GPIO15 strapped to GND via a 10k resistor or by a direct connection?
- Please isolate the culprit by assigning different pins to cs, dc, and res one after the other.

marcelstoer, 2016-08-07:
No conclusive evidence of a firmware bug and no feedback from OP in a month -> closing.

For further support: https://nodemcu.readthedocs.io/en/latest/en/support/

closed 2016-08-07 by marcelstoer