nodemcu/nodemcu-firmware #1394: MQTT disconnected, offline event not called

JosefRypacek, 2016-07-13:
### Actual behavior
Hi,
I have ESP as MQTT client. Sometimes connection is lost (MQTT broker know that client is disconnected and publish 'offline' lwt). But sometimes ESP don't trigger 'offline' event and  never gets reconnected.
If i try to call connect() again I get 'already connected' message.
When MQTT is down, http server works nice.
This is hard to simulate because it happens few times a week...

### Test code
```Lua
m = mqtt.Client(id, timeout)
m:lwt('topic', 'offline', 1, 1)
m:connect(host, port, 0, 1, nil, nil)
```
### NodeMCU version
1.5.1-master, but i got this errors on multiple dev branges over time...

### Hardware
ESP 12E

NicolSpies, 2016-07-13:
- Hi, I can confirm this behavior have experienced it more than 10 times in last 8 hours. I get the client LWT from the broker.
The offline or auto reconnect does not fire.
- I periodically test the connection by trying a dummy publish message and on failure try to reconnect and then get the "already connected" message.
- I then client.close() the connection before trying to reconnect. Then the offline event fires.
- I then destroy the client object by setting it to nil and follow the normal mqtt client creation and connection. This does not work and it is not possible to reconnect.
- If I restart the chip the mqtt connects normally again.
- The only option I have not tried is to disconnect the wifi.sta from the AP and then try to reconnect the mqtt.
- This is directly related to Issue #1395

marcelstoer, 2016-07-19:
Tracked in #1406.

closed 2016-07-19 by marcelstoer